myriadrf/LimeSuite #10: lime-suite: namespaces for public headers

guruofquality, 2015-12-17:
This might be more of a matter of taste, but since the lime-suite is largely C++, we might want to put namespace lime {} around classes and enums found in public headers. There should not be many changes to the code-base. The implementation cpp files would gain the line ```using namespace lime```, and the headers would gain the line ```namespace lime {``` and a closing bracked ```}```.

rjonaitis, 2016-01-13:
Some enumerations were clashing with CyAPI on windows, so I added lime namespace to all control logic files.

guruofquality, 2016-01-14:
Looks pretty good. We can always add more namespace lime if something is missing. I guess we can close out this issue then :-)

closed 2016-01-15 by rjonaitis