myriadrf/LimeSuite #8: New connection module: STREAM + EVB7 over USB

guruofquality, 2015-12-17:
The USBConnection will be be re-factored into a new Connection registry entry and IConnection that implements streaming and SPI using the LMS64C protocol

guruofquality, 2015-12-30:
ConnectionUSB ported over: https://github.com/limemicro/lms7suite/tree/connection_develop/src/ConnectionSTREAM

* Currently building and testing on linux. Enumeration and device access working so far. 
* The next major step will be to wrap up the streaming API and test it.
* We will probably want to re-write the CCyUSBDevice device handle stuff on windows such that multiple USB connections can be used. One CCyUSBDevice for enumerate, and then one for each individual connection.

rjonaitis, 2016-01-26:
@guruofquality made fixes for handling libusb asynchronous transfers. At the moment IConnectionIntegration branch failed to compile, so i put it into usb_timeout_fix branch 91836010ba1519c86f771a80e3fa0684fc443114

guruofquality, 2016-01-31:
I wanted to get this out of the way so we can merge this branch eventually: https://github.com/limemicro/lms7suite/commit/715653a3a758e92817361292f1b4b048643027df

This code inherits the LMS64CProtocol to reduce code changes, but really it should directly implement the transact SPI. The LMS64CProtocol should not be mentioned in the novena files. Also the UpdateExternalBandSelect() needs to be filled in.

closed 2016-03-09 by guruofquality