myriadrf/LimeSuite #9: New connection module: Novena + LMS7

guruofquality, 2015-12-17:
This issue should start by implementing the connection registry module for the Novena with spi interface only, no streaming.

However, the work for Novena LMS6 driver with streaming should be applicable to this work. Possibly using the existing FPGA driver work from LMS6, Novena has a functional streaming RX and TX channel that can be used.

9600, 2016-01-06:
Which LMS7 board would be used with Novena?

rjonaitis, 2016-01-06:
I believe it's called MyriadRF7-Novena

9600, 2016-01-06:
Ah, I wasn't aware of this. I guess it might be nice if it were called Novena-RF7, or something else that seemed to fit with the naming of the LMS6 module, "Novena-RF". Otherwise it can get a bit confusing...

guruofquality, 2016-04-10:
FYI - This is complete with the exception of UpdateExternalBandSelect which has the old implementation commented out. It needs to be reimplemented around the new API hook:
```
void ConnectionNovenaRF7::UpdateExternalBandSelect(const size_t channel, const int trfBand, const int rfePath)
{
    //TODO...
    printf("Fill in the UpdateExternalBandSelect() function for NovenaRF7!\n");
    /*
    //in case of Novena board, need to update GPIO
    if(controlPort->GetInfo().device == LMS_DEV_NOVENA)
    {
        uint16_t regValue = SPI_read(0x0706) & 0xFFF8;
        //lms_gpio2 - tx output selection:
        //		0 - TX1_A and TX1_B (Band 1),
        //		1 - TX2_A and TX2_B (Band 2)
        regValue |= Get_SPI_Reg_bits(LMS7param(SEL_BAND2_TRF)) << 2; //gpio2
        //RX active paths
        //lms_gpio0 | lms_gpio1      	RX_A		RX_B
        //  0 			0       =>  	no active path
        //  1   		0 		=>	LNAW_A  	LNAW_B
        //  0			1		=>	LNAH_A  	LNAH_B
        //  1			1		=>	LNAL_A 	 	LNAL_B
        switch(Get_SPI_Reg_bits(LMS7param(SEL_PATH_RFE)))
        {
            //set gpio1:gpio0
            case 0: regValue |= 0x0; break;
            case 1: regValue |= 0x2; break;
            case 2: regValue |= 0x3; break;
            case 3: regValue |= 0x1; break;
        }
        SPI_write(0x0706, regValue);
    }
    */
}
```

IgnasJarusevicius, 2018-06-22:
Novena support is discontinued in the current LimeSuite

closed 2018-06-22 by IgnasJarusevicius