myriadrf/LimeSuite #13: LMS7002M GUI looks terrible on Ubuntu LTS 14.04

guruofquality, 2015-12-29:
I have noticed this for a while, but I finally tried to look into it because its making the testing near impossible. It looks like wxStaticBoxSizer does not take into account the size of its label, and all of the nested box-sizers gradually loose more and more screen space the deeper they nest... just a guess. But every tab, every dialog looks like this more or less.

[image]

@rjonaitis any ideas/hints?

rjonaitis, 2015-12-29:
Which desktop environment and wxWidgets version you are using?

guruofquality, 2015-12-29:
* I tried ubuntu and kubuntu LTS 14.04 (so KDE I guess unity).
* libwxgtk-3.0-dev 3.0.0-2

rjonaitis, 2015-12-29:
The issue here seams to be related to GTK3, when wxWidgets are compiled with gtk3, the application style is affected by OS native look CSS. In this case it looks like the elements are created and layouted first, which sets their initial size, and only during rendering the css kicks in, but does not accommodate the elements size with the updated style.

Current solution would be to use wxWidgets compiled with GTK2.
Another temporary solution could be add borders to elements inside wxStaticBoxSizers to artificialy increase their size, but that would mess up the layout in other environments.

guruofquality, 2015-12-29:
You are probably right about the CSS. But at least for Ubuntu LTS, both packages libwxgtk2.8 and libwxgtk3.0 are compiled against the same GTK (2.24). So I doubt its gtk3.

rjonaitis, 2015-12-29:
libwxgtk2.8 is wxWidgets2.8, that version is too old to compile project, several defines and macros are not found in it. wxWidgets2.9 is the minimal version required to compile this project.

Did a fresh Ubuntu 14.04 LTS install.
* Compiled wxWidgets3.0.2 from source with gtk2. Compiled lms7suite, looks normal.
* Compiled wxWidgets3.0.2 from source with gtk3. Compiled lms7suite, looks normal, except scrollers background turns black.
* Installed libwxgtk3.0-dev. Compiled lms7suite, looks normal.
* Installed kubuntu-desktop, switched to KDE. launched previously compiled lms7suite, it gets distorted same as for you. Going back to Unity environment the lms7suite gets KDE style and so the distortions still appear.

So on the initial Ubuntu 14.04 LTS works good, but installing KDE environment messed something up.

guruofquality, 2015-12-29:
That was a good clue. I changed the GTK2 theme. The "gtk oxygen theme" causes the rendering problem.

[image]

It seems like oxygen theme is the default. FWIW this is the first time I have seen a problem like this. Maybe we can ignore it for now and just take note. In any case, thanks for the help.

rjonaitis, 2015-12-30:
Also one more problem is that the horizontal sliders used for parameter controls can become invisible depending on theme, it's just a minor inconvenience.
Anyway, later GUI will have to be modified to accomodate the new connection behaviour and device selections, I think this theme bug can be postponed until then.

c5e3, 2017-01-19:
I have another layout issue; at 1366x768 the window is cut off at the tab 'LDO'.
I think, a simple scrollbar would help. the windows is not scalable, so i have no change of seeing the other checkboxes
[image]

IgnasJarusevicius, 2018-06-22:
Closing old issue.

closed 2018-06-22 by IgnasJarusevicius