myriadrf/LimeSuite #14: automatic fpga loading (stream board)

guruofquality, 2016-01-02:
Automatic loading at runtime
====

For the soapy sdr bindings, we are going to want this use case to be some-what automated. This includes checking that the FPGA is loaded and that the correct image is loaded. When the driver is initialized, there where be some code to..

* check the state of the board.
* look for images on the file system,
* and to load the correct image.

Questions:
====

* Is this the latest image to use with basic streaming? https://github.com/myriadrf/STREAM/tree/master/apps/STREAM_LMS7EVB_distro_07v/gateware/bitstreams

* As an alternative, can the image be loaded into the flash: so that the STREAM board always boots into a usable image for streaming? This would replace the need for automatic loading.

* What's the best way to check that an FPGA is loaded. A flag in the device info packet for example?

rjonaitis, 2016-01-02:
>Is this the latest image to use with basic streaming? https://github.com/myriadrf/STREAM/tree/master/apps/STREAM_LMS7EVB_distro_07v/gateware/bitstreams

Yes, that image will give only IQ samples stream. The data is not packeted and don't have timestamps.

>As an alternative, can the image be loaded into the flash: so that the STREAM board always boots into a usable image for streaming? This would replace the need for automatic loading.

Yes there is programming options to load into FLASH memory, and to initiate boot from FLASH.

>What's the best way to check that an FPGA is loaded. A flag in the device info packet for example? 

For now the checking could be done by trying to read SPI registers from FPGA, Some registers should contain the loaded bitstream version. I'll have to checkup on that.

guruofquality, 2016-01-06:
> Yes, that image will give only IQ samples stream. The data is not packeted and don't have timestamps.

Oh yes. Just for basic testing.

> Yes there is programming options to load into FLASH memory, and to initiate boot from FLASH.

I think for stream board, the persistent FLASH and BOOT option makes the most sense.

I havent tried this myself. Is it just a matter of choosing the "Bitstream to flash" option? Or is there another step to configure the automatic loading when FX3 is ready?

> For now the checking could be done by trying to read SPI registers from FPGA, Some registers should contain the loaded bitstream version. I'll have to checkup on that.

Then it looks like we need an API to detect that the correct FPGA image is loaded. I know there are a lot of images out there. So maybe the streaming API will query a "feature mask" or version-check or something like that to know if can operate properly.

Also, I would like to create some kind of mechanism for STREAM where if the check fails, there is a useful error message with instructions for the user: Example A particular image url or even a command like wget the_url_to_fpga_file && burn_command.

PS thanks for the load FPGA API

rjonaitis, 2016-01-06:
>I havent tried this myself. Is it just a matter of choosing the "Bitstream to flash" option? Or is there another step to configure the automatic loading when FX3 is ready?

Yes, you only have to choose "Bitstream to flash", then it will be loaded each time after board reset.
"Bitstream to flash" only writes the data to FLASH memory, if you want to use that bitstream right after programming and without board reset, then you have to send the "boot from FLASH" command.

>Then it looks like we need an API to detect that the correct FPGA image is loaded. I know there are a lot of images out there. So maybe the streaming API will query a "feature mask" or version-check or something like that to know if can operate properly.

I checked what could be done about the FPGA image identifying. Right now there is no reliable way for distinguishing FPGA images.

>Also, I would like to create some kind of mechanism for STREAM where if the check fails, there is a useful error message with instructions for the user: Example A particular image url or even a command like wget the_url_to_fpga_file && burn_command.

A message log callback could be added to allow returning of text data from inside the IConnection.
That could be used for informing user about internal IConnection state, like the messages about bitstream checking, or used SPI frequency, or to report some connection specific problems.

```/*!
     * Callback for general information reporting
     * @param progressMsg string message to user
     */
    typedef std::function<void(const char* progressMsg)> InfoCallback;

guruofquality, 2017-01-16:
All firmware and gateware images have version checks which we are using. We can open a new issue if the need arises for finer grained feature checking.

closed 2017-01-16 by guruofquality