myriadrf/LimeSuite #15: Adding convenience calls in LMS7002M.h

guruofquality, 2016-01-03:
While making the wrapper calls for various settings for the SoapySDR wrapper. I was thinking that it would make sense if this logic was embedded into the LMS7002M.h API itself for maximum re-use. Examples:

RF switches
====

```
SetPathRFE(some enum)
{
//this call sets SEL_PATH_RFE as well as the required enables like EN_INSHSW_*
//Also it would call _conn->UpdateExternalBandSelect(...)
}

SetBandTRF(int band)
{
//band is 1 or 2
//set SEL_BAND1_TRF and SEL_BAND2_TRF
//also UpdateExternalBandSelect(...)
}
```

Gain settings
====

```
SetGainPGARBB(const double gain_dB)
{
//set G_PGA_RBB gain in dB
//another other enables or calibrations...
}

//Rx LNA, Tx PAD, etc...
```

rjonaitis, 2016-01-05:
Of course these can be added. It is planned to make a simplified version of the GUI, sort of a "wizard" for quick board setup. So this kind of functions will have to be made eventually.

guruofquality, 2016-01-05:
Cool. Any logic that wasn't in the LMS7002M.h, I have implemented in https://github.com/limemicro/lms7suite/blob/IConnectionIntegration/src/SoapyIConnection/Settings.cpp For the gains, I used the data sheet to scale a value from dB into a register value. That should probably be checked more carefully. Anyway, my API additions for LMS7002M.h would be:

* Set/GetPathRFE()
* Set/GetBandTRF()
* Set/GetPGA_RBB()
* PGA RBB MIN/MAX constants in dB
* Same for LNA RFE, RFE_TIA, TRF_PAD
* Set interpolation only (no changes to clock or rx chain)
* Set decimation only (not changes to clock or tx chain)

Ideally the Settings.cpp would be just a wrapper, without any smarts. And there will probably be a few more helper functions as well.

guruofquality, 2016-01-07:
* Added RF path settings: https://github.com/limemicro/lms7suite/commit/523c5782c390f7cc9f81262887af044c8be8cd8f

guruofquality, 2016-03-24:
Gain: https://github.com/limemicro/lms7suite/commit/fefde242b273ca76a3fa913f0ee5f5edc5fc5974

guruofquality, 2016-04-03:
Did everything but... LDO, and XBUF config are a bit advanced and hardware specific. Any custom hardware module should be able to setup these registers using the LMS7002m driver register API. So closing.

closed 2016-04-03 by guruofquality