myriadrf/LimeSuite #16: LMS7002M API and reference frequency

guruofquality, 2016-01-03:
Some calls take a reference frequency, some dont:
```
	float_type GetReferenceClk_SX(bool tx);
	float_type GetFrequencyCGEN_MHz();
        //...I believe this uses the ref freq that was last set by SetFrequencySX
	liblms7_status SetFrequencyCGEN(float_type freq_MHz);
	float_type GetFrequencySX_MHz(bool tx, float_type refClk_MHz);
	liblms7_status SetFrequencySX(bool tx, float_type freq_MHz, float_type refClk_MHz);
	liblms7_status SetNCOFrequency(bool tx, uint8_t index, float_type freq_MHz);
        ///...although this refClk_MHz is the DSP/ADC rate
	float_type GetNCOFrequency_MHz(bool tx, uint8_t index, float_type refClk_MHz, bool fromChip = true);
```

Possible changes
====

One of the things that I did to IConnection was add a call to query the reference frequency. The reference frequency is set by the board, external to LMS7002M. Some boards will have it fixed, some will have a synthesizer like stream.

I propose that all of these calls drop the reference frequency parameter, and in addition, LMS7002M does not cache the reference freq. Ref freq is always queried from one place, the IConnection.

IConnection.h
```
    /*!
     * Query the frequency of the reference clock.
     * Some implementations have a fixed reference,
     * some have a programmable synthesizer like Si5351C.
     * @return the reference clock rate in Hz
     */
    virtual double GetReferenceClockRate(void)
```

rjonaitis, 2016-01-05:
I agree, the reference clock can be moved to IConnection.

Some notes about reference clocks:
* SXR and SXT can have separate reference clocks, but it is most likely that they always will be set to identical values.
* CGEN reference clock is always equal to SXR reference clock.
* Fixed reference clock boards can depend from used crystal. There are EVB7 boards that use one of the 30.72MHz or 52MHz crystal.

Also it would be a good idea to match units in the APIs, so possibly switch from MHz to Hz.

guruofquality, 2016-01-05:
> SXR and SXT can have separate reference clocks, but it is most likely that they always will be set to identical values.

Right the XBUF config can have different clocks in some configurations. Perhaps we can have an additional set of calls

* GetTxReferenceClockRate() //without overloading, calls GetReferenceClockRate()
* SetTxReferenceClockRate()  //without overloading, calls SetTxReferenceClockRate()

> Also it would be a good idea to match units in the APIs, so possibly switch from MHz to Hz.

I have made a mistake sometimes with the Hz vs MHz. :-) I like to have absolute units when possible. Hz would be preferable in my opinion.

rjonaitis, 2016-01-06:
Since the parameters will be changed to Hz, we might as well change the frequency parameter types from float to int64_t.

guruofquality, 2016-01-07:
I dont know if using a 64-bit integer is worth the trouble. Like: we might end up with strange precision issues when representing frequencies with repeating decimals, like 33.333... MHz. And we would loose the ability to represent sub-hz frequencies, like CORDIC precision. Maybe in reality, this isnt such a problem but I would stick with double for all units (Hz, volts, dB, etc) unless efficiency required it.

guruofquality, 2016-04-03:
Use ref freq API: https://github.com/limemicro/lms7suite/commit/72c316b409912bd9d8cbe240bf9fb386b8c44338

guruofquality, 2016-04-03:
And using Hz has its own issue: https://github.com/limemicro/lms7suite/issues/39

Closing

closed 2016-04-03 by guruofquality