myriadrf/LimeSuite #17: Support multiple/simultaneous stream boards

guruofquality, 2016-01-08:
The ConnectionSTREAM should support multiple + simultaneous boards. I don't think this is a priority, but lets track it here to remember.

1) Consider the case where multiple STREAM boards are attached VIA USB. We need a way to select a specific board. It looks like currently the Cypress API supports this using the device index, but the libusb API uses a VID/PID lookup that may be ambiguous with multiple boards.

2) Consider the case using simultaneous stream boards where someone may use two STREAM connections in the same process. In addition to 1), It also looks like the Cypress API implementation shares a device pointer for all open devices. So its currently limited to one device at a time.

rjonaitis, 2016-01-08:
>1) Consider the case where multiple STREAM boards are attached VIA USB. We need a way to select a specific board. It looks like currently the Cypress API supports this using the device index, but the libusb API uses a VID/PID lookup that may be ambiguous with multiple boards.

This can be solved by using `libusb_get_device_list`, it's currently used in `ConnectionSTREAMEntry::enumerate` function to collect device PID&VID and usb speed modes. So device index can be used with libusb as well, to get device handle from that list, which would be used for opening device instead of PID&VID.

>2) Consider the case using simultaneous stream boards where someone may use two STREAM connections in the same process. In addition to 1), It also looks like the Cypress API implementation shares a device pointer for all open devices. So its currently limited to one device at a time.

Cypress API is not limited to single open device, multiple devices can be opened by creating additional `CCyUSBDevice` objects.

guruofquality, 2016-01-08:
>  So device index can be used with libusb as well, to get device handle from that list, which would be used for opening device instead of PID&VID.

Sounds good. I was thinking: Is there a unique ID associated with the stream board that we could query: like a serial number or even the mac address? An index might change when a new board is inserted, which could make the ConnectionHandle refer to the wrong device. A unique ID would prevent this.

> Cypress API is not limited to single open device, multiple devices can be opened by creating additional CCyUSBDevice objects.

Gotcha. I just meant that the current StreamConnection only uses a single CCyUSBDevice.

rjonaitis, 2016-01-08:
>Is there a unique ID associated with the stream board that we could query: like a serial number or even the mac address?

Currently there is no way of uniquely identifying boards.

guruofquality, 2016-01-08:
I see. If there was a register or field that could be read/written in the FX3 that would persist for as long as the device was powered (could be permanent); then the host could load this field with a "UUID1" -- for the same purpose. :-)

guruofquality, 2016-03-09:
Make use of the new readback ID: https://github.com/limemicro/lms7suite/issues/28

guruofquality, 2017-01-16:
Completed serial level filtering this commit: https://github.com/myriadrf/LimeSuite/commit/da9ab76b4a4697d82c8bced75bae178359f35b8b

closed 2017-01-16 by guruofquality