myriadrf/LimeSuite #18: transfer timed out/Error: popping from RX/rate: 0 kB/s failures:2

guruofquality, 2016-01-08:
@rjonaitis I'm having trouble getting useful data out of the stream board. I am really looking for a positive confirmation of IQ samples. Here are the steps that I took, and whats happening:

* This is using the **master** branch (and not the connection development work). I have a STREAM board that is attached via USB3, and EVB7 attached via COM. This EVB7 does not have the resistors populated, and the power jumper is not populated.

* This FPGA image was burned to flash: https://github.com/myriadrf/STREAM/tree/master/apps/STREAM_LMS7EVB_distro_07v/gateware/bitstreams

* After power cycle the led3 is solid and eventually goes off. I assume this means that the FPGA has loaded properly.

* launch the app, sudo bin/lms7suite

* Connect to stream and com ports

[image]

* This is the correct FW version, right?

[image]

* Open file: https://github.com/myriadrf/STREAM/blob/master/apps/STREAM_LMS7EVB_distro_07v/gui/wfm_tx_rx_61MHz.ini (no errors)

* GUI -> Chip (no errors)

* Open FFT Viewer. click start. Nothing happens for about ten seconds. Then I get some all zero IQ samples on the screen, and this verbose in the terminal. Any ideas? Thanks.

[image]

```
sudo bin/lms7suite 
Visible GUI update time: 0
GUI update time: 0
Create time 1172 ms
Claimed Interface
Visible GUI update time: 17
GUI update time: 264

(lms7suite:7120): Gtk-CRITICAL **: IA__gtk_window_resize: assertion 'width > 0' failed

(lms7suite:7120): Gtk-CRITICAL **: IA__gtk_window_resize: assertion 'width > 0' failed

(lms7suite:7120): Gtk-CRITICAL **: IA__gtk_window_resize: assertion 'width > 0' failed
        Error: popping from RX
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
        Error: popping from RX
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
        Error: popping from RX
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
        Error: popping from RX
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
        Error: popping from RX
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
        Error: popping from RX
transfer timed out
        Error: popping from RX
Rx: 0.0%         rate: 0 kB/s failures:2
        Error: popping from RX
Processing finished

```

rjonaitis, 2016-01-08:
How are you powering EVB7, through external 5V, through USB, or from STREAM board?
The power jumper should not be populated when using external 5V,
power jumper on 1-2 when powering from STREAM,
power jumper on 3-4 when powering from USB

guruofquality, 2016-01-08:
> power jumper on 3-4 when powering from USB

Thanks. I needed to do this. Previously I was using a EVB7 with registers populated and the "1-2" jumper. I switched the hardware for testing and and got this part wrong.

closed 2016-01-08 by guruofquality