myriadrf/LimeSuite #19: Very slow initialization for LMS_StreamBoard_FIFO

guruofquality, 2016-01-08:
I was testing with the LMS_StreamBoard, and I couldnt figure out why it took 10 seconds to get data. I found that this seemingly innocent block of code was the culprit. Nothing to do with threads or blocking:

```
    mRxFIFO = new LMS_StreamBoard_FIFO<SamplesPacket>(1024*2);
    mTxFIFO = new LMS_StreamBoard_FIFO<SamplesPacket>(1024*2);

///calls this which takes 10 seconds or more

            for (int i = 0; i < FIFO_length; ++i)
            {
                mElements.push_back(T());
                ++cnt;
            }

//The packet type (T) is big... but not that big...

        const static int samplesCount = 32768;
        float iqdata[samplesCount];
        int channel;

```

In addition, I have had some compiles that did not experience this delay (rearguards of master branch or connection work). Its weird... But the fix was to use a pointer instead (other changes needed to compile not shown):

```
diff --git a/src/LMS_StreamBoard/LMS_StreamBoard_FIFO.h b/src/LMS_StreamBoard/LMS_StreamBoard_FIFO.h
index 51d06cd..53805be 100644
--- a/src/LMS_StreamBoard/LMS_StreamBoard_FIFO.h
+++ b/src/LMS_StreamBoard/LMS_StreamBoard_FIFO.h
@@ -38,7 +38,7 @@ public:
         {
             for (int i = 0; i < FIFO_length; ++i)
             {
-                mElements.push_back(T());
+                mElements.push_back(new T());
                 ++cnt;
             }
             //mElements.resize(FIFO_length, T());
@@ -131,7 +131,7 @@ protected:
        uint32_t mElementsFilled;
 
     std::mutex mElementsLock; // condition variable for critical section       
-    std::vector<T> mElements;
+    std::vector<T *> mElements;
     std::condition_variable canWrite;
     std::condition_variable canRead;
 };

```

rjonaitis, 2016-01-08:
```
for (int i = 0; i < FIFO_length; ++i)
{
    mElements.push_back(T());
    ++cnt;
}
//mElements.resize(FIFO_length, T());
```
Perhaps the pushing to vector causes a lot of memory reallocations that is slowing it down. I've never encountered slowdown in that section.
Anyway, previously this was done with vector resize which should allocate all needed memory in one go, but for unknown reason it always gave segfaults, somehow changing the code to loop solved it.

guruofquality, 2016-01-08:
> mElements.resize(FIFO_length, T());

I think using the vectors causes a lot of resizes and FIFO_length! (factorial) memory copies beach each element is copied every time there is a resize. But switching the push_back with resize doesn'y fix the delay for me. :-( No idea.

Anyway, FIFO_length * samplesCount * float = 1024*2*32768*4 = 256 MiB And that could fail allocation to a single slab of memory.

closed 2016-01-08 by guruofquality