myriadrf/LimeSuite #22: Return codes, status codes, enums and more

guruofquality, 2016-01-14:
So there are a few different error enums, to name a few:

* liblms7_status
* OperationStatus
* eCMD_STATUS

I was thinking that it might be cleaner to follow a different model for dealing with error codes and error reporting. Basically, there would be one error reporting and querying API. And most API calls would simply return pass/fail true/false 0/-1. The user would be able to query the actual error code and message string when the call failed, and if they were interested. Its a lot like errno or windows getlasterror().

Possible API
====

**list of pre-defined error codes** that users will encounter when using lime suite
```
SUCCESS = 0
ERROR_CONNECTION,
ERROR_RANGE
ERROR_BUFFER
etc...
```

When API calls fail, they should first **report the error code**
and an optional message string with more details
```
void ReportError(int code, string message = "");

//variant for reporting after system call failures that use errno/getlasterror
void ReportError(int code, errno);
```

When a user wants to **query the error**:
```
//get the error code reported
int GetLastError(void);

//get the error code to string + any optional message reported
const char *GetLastErrorMessage(void);
```

Example
====

The LMS API call CalibrateTx():
```
bool LMS7002M::CalibrateTx(float_type bandwidth_MHz)
{
    if (bandwidth_MHz <= 0.0)
    {
        lime::ReportError(ERROR_RANGE, "LMS7002M::CalibrateTx() expects BW > 0");
        return false;
    }

    return true;
}

```

The user's code:
```
if (not rfic->CalibrateTx(-100))
{
    std::cout << "Failed to calibrate! " << lime::GetLastErrorMessage() << std::endl;
}
```

Notes
====

The error reporting API is global, its not tied to any particular class or instance. The error reporting API will store error codes and messages in thread-local-storage to make it thread-safe. As long as a user queries the error after a failure before making a new lime-suite API call, there will never be a problem.

rjonaitis, 2016-01-15:
Looks good, we can definitely use this.

guruofquality, 2016-03-24:
* https://github.com/limemicro/lms7suite/commit/f98ca33803a136e8a58b93e28ecfb9403a9ac9c9

I added the set gain calls with error code return -- while I was adding it. But I guess the real first step is to do this re-factor to the IConnection and LMS7002M classes.

guruofquality, 2016-03-30:
Work on this branch - https://github.com/limemicro/lms7suite/tree/ErrorCodes

It touches quite a bit of code, so I think this will take some more time, and we shouldn't let it stall other issues.

guruofquality, 2016-04-05:
Updated info...

* The new API is implemented: https://github.com/limemicro/lms7suite/blob/master/src/ErrorReporting.h
* Work has started on this branch: https://github.com/limemicro/lms7suite/commits/ErrorCodes

**API call should look like this**
```
#include <ErrorReporting.h>

int myCall(const int arg0, const std::string &arg1)
{
  if (arg0 != 3)
  {
    //calls that exit with error can return ReportError with a formatted message
    //The ReportError API also accepts standard error codes or errno when applicable.
    //ReportError() returns -1, other codes can be used with an explicit return after ReportError()
    return ReportError("myCall got bad input (%d, %s)", arg0, arg1.c_str());

   //its also possible to indicate standard error codes
   return ReportError(EIO, "myCall got bad input");
  }
  return 0; //success!
}
```

guruofquality, 2016-04-11:
Some files still have their own SUCCESS/FAILURE enums, but all major interfaces switched over to this. Classes like LMS64C protocol and LMS7002 driver should have better reporting now. In addition, some connection classes had commented out errors that are now reported to the interface.

closed 2016-04-11 by guruofquality