myriadrf/LimeSuite #24: STREAM: support sample rates below 2.5 MSps

guruofquality, 2016-02-15:
Due to the limitations of the PLL in the fpga, the minimum supported sample rate is 2.5 MSps or 5 MHz PLL. Some applications (like GSM) and user testing may require lower sample rates.

One suggested fix may be to use a clock mux to switch between the PLL and an alternative clock for lower rates. A slower clock with appropriate phase relationship could be generated in register logic by oversampling the MCLK with a faster clock and creating the required phase delay.

guruofquality, 2016-03-09:
* Note: implemented in latest FPGA image, needs a software hook in the set PLL rate hook

rjonaitis, 2016-03-10:
SoDeRa gateware and firmware updated at 5e0fe51, when setting sampling rate below 2.5 Msps, sampling rate is switched to 0.541666 Msps

9600, 2016-03-10:
Does this mean that the sampling rate can be arbitrary above 2.5Msps and up to the upper limit, but below this it is fixed at 0.541666 Msps? Therefore, e.g. 2Msps, would not be possible?

rjonaitis, 2016-03-10:
Yes, that is correct

9600, 2016-03-10:
@guruofquality what would this mean for apps such as the BTLE demo that uses a 2M sample rate?

Would another option not be to have configurable decimation/interpolation in the FPGA and run the converters at higher rates (above 2.5 Msps)? Or would that be a lot of work or maybe require too much resource in the FPGA?

guruofquality, 2016-03-10:
So I don't have a good grasp on what the PLLs are used for and what other technical barriers this Altera FPGA imposes.

I may have mentioned this before, but I can say that I solved this problem in the past on a series7 Zynq Xilinx FPGA doing the following: The RX MCLK clock is captured directly into a clock buffer and used to clock FPGA cores for LML capture with IDDR and for FIFOs in that clock domain. The TX MCLK is captured directly into a clock buffer and used to clock FPGA cores for LML generation with ODR and for FIFOs in that clock domain. There is however a single PLL used to generate the TX FCLK (feedback clock) which needs a 90 degree phase shift from the original TX MCLK. There is no RX PLL because there is no RX feedback clock required for the LML interface in this case.

So, when the TX MCLK sample rate becomes too low for the PLL to handle, the TX FCLK is swapped out with a clock mux for an oversampled clock. To generate the oversampled clock, the TX MCLK is sampled with a faster clock (in the 10s of MHz range) and the transitions are detected to generate the same 90 degree phase shift.

So, I don't recommend decimation/interpolation in the FPGA if the above is a possibility. But, I strongly think that this board needs to have a full range of flexible sample rates available to the user. I'm not very familiar with Altera, but I am curious to hear if this is a technical impossibility for Altera clocking.

closed 2016-04-18 by rjonaitis