myriadrf/LimeSuite #25: STREAM: TX IQ random swap issue

guruofquality, 2016-02-15:
The IQ can random swap on TX between runs. This might be an issue with GPIFIII state machine. Uncertain. To reproduce, run, ctrl+c, and re-run the siggen app. The tone should be positive of the TX LO, but will sometimes swap positions:

```
python lms7suite/src/SoapyLMS7/siggen_test.py \
    --clockRate=40e6 --txAnt=BAND1 --txGain=52 \
    --args="driver=lime" --freq=1000.5e6 \
    --ampl=0.7 --rate=5e6 --waveFreq=300e3
```

rjonaitis, 2016-02-17:
The issue is related to LMS_StreamBoard::ConfigurePLL(). Disabling it after FPGA PLL is configured correctly, does not make random IQ swapping anymore.
I don't know the exact reason why that's happening, perhaps PLL reset is needed after all those rapid ConfigurePLL calls.

guruofquality, 2016-03-09:
Definitively still seeing this. Whatever the state machine is, it needs a reset or some kind of enable line to disable during calibration and/or PLL changes.

ztamosevicius, 2016-04-04:
The issue arises when FPGA PLL are configured while LMS7002M CGEN is not configured yet. Then FPGA doesn't get reference clock from MCLK lines and FPGA PLL configuration is messed-up.

Resolution: LMS7002M CGEN must be configured and reference clocks (i.e. MCLK1/MCLK2) must be provided before FPGA PLL configuration.

guruofquality, 2016-04-04:
I think that then the PLLs are configured, that CGEN has always been configured, and both mclks have something on them. I guess that I am concerned about metastability.

What happens when the CGEN clock rate is changed after the PLL configure? In this case, the mclks will go down temporarily, and change rates. Then the  configure PLL call is needed again. How should we handle that case properly?

rjonaitis, 2016-04-04:
New gateware will have FPGA PLL reset, that should solve the problem

guruofquality, 2017-01-16:
No present issues. Closing.

closed 2017-01-16 by guruofquality