myriadrf/LimeSuite #26: osmotrx, timestamps, and non-monotonic

guruofquality, 2016-02-15:
Because of the re-calibrations, the streaming is disabled and enabled. This may be causing the timestamp to reset to 0. One possible fix would be to copy the last timestamp into the timestamp offset. This will make the timestamps appear monotonic from the API perspective:

Example, although this did not immediately work (caused application to exit):
```
diff --git a/src/ConnectionSTREAM/ConnectionSTREAMing.cpp b/src/ConnectionSTREAM/ConnectionSTREAMing.cpp
index f9a9ae1..826eabd 100644
--- a/src/ConnectionSTREAM/ConnectionSTREAMing.cpp
+++ b/src/ConnectionSTREAM/ConnectionSTREAMing.cpp
@@ -158,6 +158,10 @@ struct USBStreamService : StreamerLTE
         //clear any residual data from FIFO
         ResetUSBFIFO(dynamic_cast<LMS64CProtocol *>(mDataPort));
 
+        //hardware time resets to zero, correct with offset + 100 ms
+        mTimestampOffset += mLastRxTimestamp + mHwCounterRate*0.1;
+        mLastRxTimestamp = 0;
+
         //switch on Rx
         auto regVal = Reg_read(mDataPort, 0x0005);
         int syncDis = txTimeEnabled?0:(1 << 5);

```

rjonaitis, 2016-02-17:
Added updated SoDeRa firmware 71737b3fe, now it does not reset timestamp when enabling/disabling stream. Timestamp can be reset to 0 manually.

9600, 2016-03-04:
@guruofquality is this now fixed?

guruofquality, 2016-04-04:
I think we got this. I didnt see the error.

closed 2016-04-04 by guruofquality