myriadrf/LimeSuite #28: STREAM: flash unique ID for each board

guruofquality, 2016-02-15:
Just an idea, but perhaps with a firmware update, a UUID could be read/written to the FX3 flash. When the cal software sees an uninitialized UUID, it would generate a random one and write it once. This would give us a unique ID for each hardware in the future for calibration data.

rjonaitis, 2016-02-16:
32 bit board serial number will be available to read using GET_INFO command.

guruofquality, 2016-03-09:
We can use this ID to make calibration tables unique. And also to list in device enumeration serial number and selecting a specific device using the connection handle.

guruofquality, 2016-04-09:
Additional TODO: modify ConnectionSTREAMEntry.cpp to use the serial in the ConnectionHandle::serial field. This will let users differentiate between multiple boards using the serial number in the device handle. It looks like we will have to open the connection with an instance of ConnectionSTREAM() to call GetDeviceInfo().

guruofquality, 2017-01-16:
The serial field is populated from libusb_get_string_descriptor_ascii() as reported by the FX3 firmware.

closed 2017-01-16 by guruofquality