myriadrf/LimeSuite #30: Does not support use with lower frequencies.

9600, 2016-03-04:
E.g.

$ osmocom_fft -F -a "driver=lime,soapy=0,clock=40e6,cacheCalibrations=1" -s 5e6 -f 7e6

Results in:

LMS_StreamBoard::ConfigurePLL(tx=10MHz, rx=5MHz)
C[INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 7.000000 MHz), ref 30.720000 MHz
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, -1897.655000 MHz), ref 30.720000 MHz

See also attached for FFT frequency scale displayed.
[image]

guruofquality, 2016-03-04:
>> So I think SetFrequencySX() or GetFrequencySX_MHz() didn't handle 7MHz correctly.

> Correct, these functions configure only Rx/Tx PLL frequencies, and they
can't be configured that low. To achieve 7 MHz frequency additional chip
configuring is required. Perhaps a higher level function should be created
to handle all that.

So I think the call to set the RF frequency needs to clip the requested frequency to the closest possible frequency. When that happens, the default tuning algorithm in SoapySDR will automatically compensate for the error using the CORDIC.

If for some reason we don't like that automatic behaviour with the clipping and cordic correction outside of the driver, then SoapySDR::setFrequency(dir, chan, freq){} can be overloaded to implement something smart as well.

Is that basically the right idea here? And what is the actual/achievable lower bound that the LO can tune?

9600, 2016-03-05:
@guruofquality are you suggesting that applications would have to do something different with this particular hardware at lower frequencies? That doesn't sound right.

@rjonaitis could you comment on LO bounds.

guruofquality, 2016-03-05:
> @guruofquality are you suggesting that applications would have to do something different with this particular hardware at lower frequencies? That doesn't sound right.

Nope, just driver related work in lms7suite. The SoapyLime::setFrequency() call will just need to clip the target frequency to a possible frequency.

It probably gets a bit hairy to tune near the lower VCO band edge, so we might need a slightly conservative lower bound, and I think it might be a function of the clock rate.

guruofquality, 2016-03-06:
FWIW, I have been using fixed (100e3, 3.8e9) as the tunable LO range in the LMS7002M. I may have pulled that from the data sheet rather than testing edge cases. I also added this as an API call to add to the LMS700M class, so if its configurable, that can be added like getSXXRange(): https://github.com/limemicro/lms7suite/issues/15

rjonaitis, 2016-03-07:
> could you comment on LO bounds.

PLLs can be tuned from around 30 MHz to 3.8 GHz

guruofquality, 2016-03-10:
This should do the trick: https://github.com/limemicro/lms7suite/commit/27b488b75bfb55339f828f120b8617d50c57de8b

closed 2016-03-24 by guruofquality