myriadrf/LimeSuite #32: Transmit stops working after pause in stream

guruofquality, 2016-03-09:
Whenever there is a pause in the transmit stream, accidental/underflow or forced/end of burst, the transmit samples seem to have no effect. It looks like we just see the TX LO at whatever level was previously the last sample. The following example sleeps every 3 seconds for 3 seconds, it should cause a tone to go on an off. Once the tone appears, it goes away forever. If the zero-flush at the end is removed, the LO can be seen at full power (possibly because the last tx sample before the sleep was at full amplitude)

* demo code checked in: https://github.com/limemicro/lms7suite/commit/d478333c4c2a492e24cd072e1529dad55e8c6aa3

```python demo_tx_recovery_issue.py --clockRate=32e6 --txAnt=BAND1 --txGain=40 --args="driver=lime" --freq=433.92e6 --ampl=0.7 --rate=4e6 --waveFreq=300e3```

rjonaitis, 2016-03-10:
SoDeRa firmware and gateware updated at 5e0fe51, stream now is recovering after pauses.

guruofquality, 2016-03-18:
Works for me, thanks.

closed 2016-03-18 by guruofquality