myriadrf/LimeSuite #34: merge Iconnection work into master

guruofquality, 2016-03-18:
Can we merge Iconnection branch, and should we merge before things diverge any more? My concerns are:

* novena has diverged in master branch
* possible bugs introduced in Iconnection

rjonaitis, 2016-03-18:
Currently IConnection branch does not support separate Stream + EVB7 connections and has not been tested on Windows OS. 
Once this is done, it could be merged.

Alternatively right now master could be merged into IConnectionIntegration

guruofquality, 2016-03-21:
> Currently IConnection branch does not support separate Stream + EVB7 connections and has not been tested on Windows OS. 

You mean EVB7 COM port + Stream USB? This is not supported under a single IConnection, but I believe this does work for the GUI. And so I dont think anything in missing or broken in this regard.

> Alternatively right now master could be merged into IConnectionIntegration

Thats probably the best, merge master into IConnectionIntegration and test. Once IConnectionIntegration fully works, it can be merged back into master.

rjonaitis, 2016-03-21:
> You mean EVB7 COM port + Stream USB? This is not supported under a single IConnection, but I believe this does work for the GUI. And so I dont think anything in missing or broken in this regard.

Yes, the GUI still has two separate connections, so it's working ok on Linux. Have not yet tested it on Windows.

master branch has been merged into IConnectionIntegration

guruofquality, 2016-03-30:
@rjonaitis what do you think about merging this work into master now? I don't think there will be much chaos. I'm pretty confident that if we find a bug we can fix it quickly. And everyone seems excited to make the new connection work mainline.

rjonaitis, 2016-03-31:
IConnectionIntegration has been merged to master branch

closed 2016-03-31 by rjonaitis