myriadrf/LimeSuite #35: libusbx warnings.

9600, 2016-03-20:
Frequently get libusbx warnings of the sort:

2016-03-20 12:34:15 ::1: libusbx: warning [handle_timeout] async cancel failed -5 errno=22

E.g. when using PothosGui and changing settings or deactivating a topology.

guruofquality, 2016-03-21:
I think these are prints from when the worker threads are stopped. There are actually more prints, but they are commented out. If we ever isolate the worker threads from the calibration, then you will see these a lot less -- like once at shutdown.

guruofquality, 2016-03-21:
Another data point: After stopping the topology, I also saw this print continually. In this case the rx thread is still running, which is probably the source. But not sure why. It didn't affect anything from bring re-activated properly.

commit 77faf477 (2016-03-24): Fixes #35, move libusb events handling to dedicated thread
closed 2016-03-24 by rjonaitis