myriadrf/LimeSuite #40: Thoughts for baseband filter cal speedup

guruofquality, 2016-04-04:
**Read modify write should use fromChip = false**

We have a cache of LMS700m registers, so it should be possible to specify fromChip false. This should cut down the operation time of all Modify_SPI_Reg_bits() used in the filter setup.

**Use bulk writes during cal setup**

Rather than writing out to the LMS7 registers on each call to Modify_SPI_Reg_bits(), we could also implement a toChip  = false for Modify_SPI_Reg_bits(). Then perform a single bulk write of all affected registers.

**Calibration database for RC cal values**

Store all calibration results in the database. Unless the user is sweeping over possible bw settings, we are likely to see the same values come up again and again. 1) an initial value set by the driver that is good for most applications, and 2) an application specific value that is used over and over.

**Combine RX filter TIA (RFE) and LPFL/H (RBB) calibration**

Rather than a separate cal operation for both RFE and RBB, (and only when the requested BW is the same for both filters)... most of the calibration uses the same setup (same clock rate, same loopback, same LO). One operation with a shared initial setup should be faster.

rjonaitis, 2016-04-04:
> Read modify write should use fromChip = false

Yes, that is a valid point. Although care should be taken when using values from register cache, writing to some registers can affect values in other registers and the cache will return values different than those in the chip.

> Use bulk writes during cal setup

Yes, it's possible to make a batch of registers for the setup. Although there are some registers where writing order matters, so their end result can't be just written once. Need to watch out for those. Also the setup might need some changes, so for now it's more convenient to have separate Modify_SPI commands for editing, than to rewrite them into batches of register addresses and values.
Earlier I was thinking about adding additional functions like BeginBatch(), EndBatch() which would allow to group existing functions into separate batches in respect to channel changes and writing order.

> Calibration database for RC cal values

Database is already made for other calibrations, it should be sufficient just to add another table for filter values to it.

guruofquality, 2016-04-08:
Update, I switched to fromChip = false as the default: https://github.com/limemicro/lms7suite/commit/bfb282d7048e7b3dd0e6cea58e1125147ba2dc91 I made the effort to explicitly use fromChip = true in calls that required readback from chip (vco compare, rssi).

If there is a problem, this is a simple commit to revert. In any case the speedup is noticeable and appreciated while debugging. Still looking forward to faster SPI implementation :-)

rjonaitis, 2016-05-12:
Filter calibration algorithms have been updated. RX filter TIA (RFE) and LPFL/H (RBB) calibrations joined into single procedure. Currently it does not use results caching, it can be added later.

guruofquality, 2016-05-17:
@rjonaitis I added caching versions of the calibration routines to master. It was easier to implement them as separate functions than to pollute the current cal functions. I hope this way is better: https://github.com/myriadrf/LimeSuite/commit/9bf3c8e6c6c69274fab6ee89dff1bb05435360e2

Closing this issue since all the major things were taken care of. Lets open a new issue if something comes up. Thanks.

closed 2016-05-17 by guruofquality