myriadrf/LimeSuite #45: LMS7 Suite GUI - OSX rendering issue

guruofquality, 2016-05-09:
It looks like only the first tab of the gui renders. Calibration is fine, but RFE, RBB, etc.. are all empty when selected. Could be a resizing hook issue -- maybe we need to force update/resize after tab changes.

Screenshot for reference in this thread, only cal tab is shown: https://discourse.myriadrf.org/t/is-limesdr-mac-compatible/329

rjonaitis, 2016-05-10:
I don't have OSX system, so i can't test it myself, but i suspect that lms7002_mainPanel::Onnotebook_modulesPageChanged event function might override native event responsible for redrawing.

adding repaint event should possibly fix that
```
--- a/src/lms7002_wxgui/lms7002_mainPanel.cpp
+++ b/src/lms7002_wxgui/lms7002_mainPanel.cpp
@@ -259,6 +259,7 @@ void lms7002_mainPanel::Onnotebook_modulesPageChanged( wxNotebookEvent& event )
         rbChannelB->Enable();
     }
     UpdateVisiblePanel();
+    Refresh();
 }
```

guruofquality, 2016-05-10:
Thanks for the hint. This fixed it: https://github.com/myriadrf/LimeSuite/commit/a9cda60830efbad8f91affd0f9b6878d395a34ed

closed 2016-05-10 by guruofquality