myriadrf/LimeSuite #46: Refactor fftviewer around IConnection streaming API

guruofquality, 2016-05-09:
Currently fftviewer is implemented around LMS_StreamBoard API and StreamerNovena API. But it should be generic for any board supporting streaming under IConnection.

* In the case of LMS_StreamBoard, IConnection should already support streaming.
* In the case of StreamerNovena, these functions belong inside of ConnectionNovenaRF7, some re-factoring is required for ConnectionNovenaRF7 to implement the stream hooks.

Thoughts?

rjonaitis, 2016-05-10:
It's pretty straightforward to update fftviewer to use IConnection.
But the Setup/ControlStream implementations have to be reworked, as it now has couple of nuances: threads can't be stopped, SetHardwareTimestamp() must be called prior Setup to create the stream service, otherwise it segfaults... So the fftviewer won't work correctly in current situation.

What should be done:
- [ ] Move Rx/Tx threads from StreamerLTE to each IConnection class
- [ ] Move FFT calculations thread from StreamerLTE to FFT GUI
- [ ] Remove StreamerLTE, without the threads it has no more purpose
- [ ] Update Setup/ControlStream functions to work correctly

The way i see it:
- SetupStream should fully configure expected samples formats and allocate memory buffers.
- ControlStream should start/stop the threads
- CloseStream should deallocate memory

Regarding the stream commands, i think there should be only one active command which could be replaced at any time, instead of commands queue.
Also does SoapySDR switch between continuous and burst modes during operation? As it is now if continuous mode fills up Rx buffer and then switch to burst mode, the samples returned from ReadStream() is not guaranteed to contain only the burst samples, there is possibility for leftover samples from continuous mode. I think when stream receives command for burst mode, it should clear all existing samples from buffers.

guruofquality, 2016-05-11:
> Move Rx/Tx threads from StreamerLTE to each IConnection class

If you mean ConnectionSTREAM class, then yes. Definitely.

> Move FFT calculations thread from StreamerLTE to FFT GUI

Sounds like the right thing as well.

> Remove StreamerLTE, without the threads it has no more purpose

Cool. That means more hardware specific classes inside of the connection implementations -- agree

> Update Setup/ControlStream functions to work correctly

You are right about the threads. thats how it should work

----
**The current timestamp problem**

The main motivation for the active receive thread was to always have a current timestamp. I can see how things would be drastically cleaner if it wasnt for this. We still need a way to get the current timestamp for certain apps, even when streaming is not setup. Can I suggest something hack-ish?

Lets refactor the code the way you suggested. We know the current timestamp whenever the rx thread is running, but we only run the rx thread intentionally after SetupStream(). When the user calls the getHardwareTime hook and there is no thread, lets just do a quick setup + teardown with SetupStream/CloseStream to get the latest time. It should do the trick. And maybe in a future FPGA rev, we can have a readback register for this.

----
> Regarding the stream commands, i think there should be only one active command which could be replaced at any time, instead of commands queue.

Just to be clear, the behaviour that I am trying to emulate is requesting specific rx samples -- this usually means continuous streaming, or multiple bursts, or specific times for continuous or bursts. Ideally the FPGA hardware would support sending only the samples requested and the rx thread would have no such concept of command queue. That way we don't stream extra samples just to throw them away inside of the rx thread.

Since there isnt hardware support, we do need some kind of software handling to wait for a specific timestamp and then enter continuous streaming. -- osmobts needs this for sure. But the queue is not very useful for continuous streaming as we would only ever see a depth of 1. But, I think there is a case for queuing up timed-bursts. Each command is completed before the next one is executed.

Because we need to soft emulate this behaviour, and cross thread boundaries, and because there could be a chain of burst commands -- I think a queue is appropriate.

----
> Also does SoapySDR switch between continuous and burst modes during operation? As it is now if continuous mode fills up Rx buffer and then switch to burst mode, the samples returned from [...]

Technically one could switch between continuous and burst modes with the API. You probably would not see this. But about the leftofver samples -- you do see this even with just starting and stopping the stream. Generally, I have suggested when one stops continuous streaming that they flush the interface by manually calling readSamples() until the call has a timeout.

I often see flushing when the stream is stopped, or at a higher level like in OsmoBTS, or in the gnuradio start() hook for an sdr block. I think it would be acceptable to flush if the stream command changed from a continuous stream to a burst, but I doubt many would even do that in the same stream instance.

guruofquality, 2016-05-12:
@rjonaitis I think we might have a few edge cases, but otherwise its a good plan. I would like to try and firstly take care of the following tasks you mentioned:

* Move Rx/Tx threads from StreamerLTE to each IConnection class
* Update Setup/ControlStream functions to work correctly

I am still befuddled by the transmit time loopback issue and I think this refactor and simplification will help. I don't want to step on any toes. So let me know if this sounds good. Thanks!

rjonaitis, 2016-05-12:
@guruofquality Sure, I actually had started working on this, but currently have other things to take care of. Take a look at 2f80c3c3283e1b9d5, maybe this will be usefull, it's work in progress so some of code is commented out

rjonaitis, 2016-07-11:
Initial streaming rework 5ff05abcf work in progress from LMS_API branch

Currently only the ConnectionSTREAM has been updated, but other connections will have the same scheme. Now the samples receiving and transmitting functions are contained within connection object.

Option to generate samples has been added to prevent other software timeouts during chip calibration, also now the threads don't need to be restarted #31.

> The current timestamp problem
>We still need a way to get the current timestamp for certain apps, even when streaming is not setup

If stream is not running, timestamp can be just reset to 0, or if it has been previously running just return the last received timestamp. Unless the app requires that the timestamp is monotonic and always increasing during whole app lifespan.

> Just to be clear, the behaviour that I am trying to emulate is requesting specific rx samples -- this usually means continuous streaming, or multiple bursts, or specific times for continuous or bursts.

The bursts command queue has been removed from the receive thread, I think it can be replaced by the stream reading function, it would just block and continuously pop samples from FIFO until desired timestamp arrives and only then return the requested samples.

StreamSetup has been changed to create single channel FIFOs, so now stream reading functions return only one channel's data per call. FIFO will be updated to always insert newest samples and discard oldest.

@guruofquality thoughts?

guruofquality, 2016-07-11:
> Option to generate samples has been added to prevent other software timeouts during chip calibration, also now the threads don't need to be restarted #31.

Cool

> If stream is not running, timestamp can be just reset to 0, or if it has been previously running just return the last received timestamp. Unless the app requires that the timestamp is monotonic and always increasing during whole app lifespan.

Basically, as long as the rx stream had been created, then there is a thread running and the latest time can be queried from the API. I think most use cases are going to be able to satisfy this requirement of having called IConnection::SetupStream() before using time.

I have some worries, but maybe we can handle that on a case by case basis. If there is ever an opportunity to add a register-based reback for the current time counter in the FPGA, then we dont have to worry about these edge cases.

> The bursts command queue has been removed from the receive thread, I think it can be replaced by the stream reading function, it would just block and continuously pop samples from FIFO until desired timestamp arrives and only then return the requested samples.

I guess the downside is that the thread has to perform the unpacking into the fifo, and then we throw out the samples in the readstream call until IConnection::ControlStream() configures it to receive something.

Just for another data point. I think bladerf runs a single receive thread and the equivalent read stream call reads until it gets the desired timestamp, just like you have suggested. The Soapybladef implements a queue in the readStream layer. The though behind the queue is that a separate control thread can "program" a set of stream request instructions without having to interact with the readStream thread.

In regards to the queue. It could be created in the SoapyLMS7::readStream layer, within the StreamConnection::ReadStream, or removed entirely -- like a queue of length 1 or a single action to perform. I think in practice most apps are going to not need the queue depth > 1, maybe queuing can wait until its needed for a specific use case.

> StreamSetup has been changed to create single channel FIFOs, so now stream reading functions return only one channel's data per call. FIFO will be updated to always insert newest samples and discard oldest.

Is that going to hurt the MIMO streaming use case? The multiple channels per stream call was to get MIMO alignment. With separate read stream calls per channel, its possible to recreate alignment using the timestamps, but perhaps not as convenient/efficient?

rjonaitis, 2016-07-12:
>Is that going to hurt the MIMO streaming use case?

It shouldn't, as long as the CPU is able to keep up with realtime data rate and the FIFOs are not fully filled. Reading multiple channels with the same thread will return aligned timestamps.

rjonaitis, 2016-07-13:
Updated SoapyLMS7 wrapper with the new streaming, siggen_test.py and LimeSuiteCalibrate.py appears to be running correctly.

guruofquality, 2016-07-18:
@rjonaitis thanks, I have been working with a user to test this. We ran into an issue that only happens on an arm platform, not PC. It seems that ConnectionSTREAM::ReceivePacketsLoop() is not exiting when ControlStream calls stop, and UpdateThreads(). Its getting stuck in a join. Adding prints to ReceivePacketsLoop can fix the issue, so there is probably some race or similar condition to fix. If you have any suggesting, let me know. We are trying to get to the bottom of this, but otherwise so far so good.

rjonaitis, 2016-07-19:
> Update, its getting stuck in WaitForReading()

In that case it can only get stuck while locking the transferLock
https://github.com/myriadrf/LimeSuite/blob/IConnection_StreamingRework/src/ConnectionSTREAM/ConnectionSTREAM.cpp#L411

You could try to remove the mutex and conditional variable, and it should work, but it will use more CPU polling the atomic contexts[contextHandle].done.load() variable in a loop.

closed 2016-09-28 by rjonaitis