myriadrf/LimeSuite #48: Replace ini parse submodule with subtree

guruofquality, 2016-05-11:
Git subtree attempts to copy or squash another git repo -- rather than submodule which is more like a lightweight symlink. Should we consider replacing the ini parse submodule with subtree instead to

* eliminate some confusion when building from source (forgetting to run git submodule update)
* ensure that tarballs from github release tags are build-able (cant use git from a tarball)

I like submodules a lot, but since this is the only case so far, it may be easier to go with subtree.

guruofquality, 2016-05-12:
As an example, I did this for test on the reorganization branch in external/ directory: https://github.com/myriadrf/LimeSuite/tree/ReorgWork

rjonaitis, 2016-05-12:
I agree, that would be more convenient for simple users

closed 2016-05-16 by guruofquality