myriadrf/LimeSuite #51: limesuite directory reorganization

guruofquality, 2016-05-11:
I'm trying to come up with a more obvious directory structure to handle growing for future board support and becoming easier for new users to discover the code. There is more than one way to handle this (obviously) so I would like to make a discussion out of it. For the most part I want to think about how to squash src/ into the top level directory in an organized manner.

**From the top level directory:**

* cmake/ - cmake find scripts and LimeSuite package files
* debian/ - debian packaging structure
* docs/ - build docs, etc
* mcu_program/ - lms7 MCU software
* udev-rules/ - udev linux rules for all supported products
* sqlite-3.12/ - internal sqlite copy
* include/lime/ - public headers that will get installed. This should be the same organization as the install so its clear to users/developers. This includes the new LMS_API work, connection registry interface, lms7002m c++ driver api, error codes, etc -- anything in LimeSuite library.
* lib/ - C++ sources for limesuite library, private header files, LMS_API C++ sources
* boards/ - all Stream/Novena/Com connection classes + lms64C protocol. These are organized separate from the lib/ sources because they are decoupled -- however they will still be built into the limesuite library as they are now.
  * ConnectionStream/
  * ConnectionNovena7/
  * ConnectionEVB7Com/
  * LMS64CProtocol/
  * others?
* lms7gui/ - wx widgets graphical utility for lms7 (looks a lot like current src/ without the public headers and limesuite library sources)

**Some problems**

Directories like Si5351C/, ADF4002/, and others are well organized with header, source, and wx support. In this alternative organization, these directories might get split up. Alternatively, they could be kept in the same directory structure under boards/ -- some deviation from the proposed directory structure could be just fine :-)

guruofquality, 2016-05-12:
I made an attempt at this reorganization here: https://github.com/myriadrf/LimeSuite/tree/ReorgWork

* I didnt update the build, so it can't be built yet. I like the organization better, even if its not perfect (some directories have mixed sources, includes, gui stuff).
* I put cpp-ini parser and sqlite into external/
* boards/ contain board specific files, and I expect some of this to squash down when we do the IConnection streaming support for novena and combine the LTE stream support (#46)
* lib/ contains general limesuite library sources and mostly retains the organization before minus the gui stuff and some obvious public headers

@rjonaitis, @ztamosevicius Let me know what you think -- I'm looking forward to this cleanup and #46 efactor

guruofquality, 2016-05-15:
Update: I merged some more 'minor' reorg work into master.

* There is a top level CMakeLists.txt, no more building from src subdirectory
* Moved cmake/ and SoapyLMS7 top top
* Moved sqlite and ini parser to external/
* Otherwise src/ is basically unchanged

IgnasJarusevicius, 2018-06-22:
Current LimeSuite directory layout seems OK.

closed 2018-06-22 by IgnasJarusevicius