myriadrf/LimeSuite #52: LMS7002M::TuneTxFilter(30e6) error

guruofquality, 2016-05-12:
I'm getting this error when using the highband:

``[ERROR] setBandwidth(Tx, 1, 30 MHz) Failed - Filter calibrated. Filter order-4th, filter bandwidth set to 30 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active``

It looks to me like there is no escaping this error when calling TuneTxFilter() above 16 MHz. Perhaps there is a missing return or goto statement. Thoughts?

rjonaitis, 2016-05-12:
Not an error, just a misuse of ReportError() to display a message. Should be replaced by other function to display notification.

As a side note what is the expected use case of:
`int lime::ReportError(const char *format, ...)`
as it returns 0, which indicates success?

guruofquality, 2016-05-12:
I would just return 0 (and not call ``ReportError``) in this case and call the logger with ``LMS7002M::Log(LOG_ERROR/WARN, "msg")``. Basically ReportError() stashes the error message for the caller code to report, either to the logger or the gui in most cases. And the logger is also configured to be forwarded to the GUI or some other logger destination.

guruofquality, 2016-05-17:
* Using the logger: https://github.com/myriadrf/LimeSuite/commit/87306af944bbdce9bf3ec9c98fe56982b2a807a6 -- reporting as INFO
* Added printf style support for Log()
* Note to self: considering API-wide logger outside of LMS7002M class

closed 2016-05-17 by guruofquality