myriadrf/LimeSuite #54: Filter calibration problems + sweep app

guruofquality, 2016-05-18:
I made a test app on the "SweepBandwidth" branch to reveal some issues setting some of the available bandwidth filter settings:

RX direction
====

``python SoapyLMS7/test_sweep_bandwidths.py --step=0.5e6 --start=1e6 --stop=60e6 --chan=0 --dir=RX``

 * 1 MHz - RxLPF frequency out of range, available range from 0.7 to 65 MHz
 * 1.5 MHz - CFB_TIA_RFE step size out of range
 * 2 MHz - C_CTL_LPFL_RBB step size out of range
 * 2.5 MHz - C_CTL_LPFL_RBB step size out of range
 * 3 MHz - CFB_TIA_RFE step size out of range
 * 3.5 MHz - CFB_TIA_RFE step size out of range
 * 22 MHz - C_CTL_LPFL_RBB step size out of range
 * 31 MHz - C_CTL_LPFH_RBB step size out of range
 * 31.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 32 MHz - C_CTL_LPFH_RBB step size out of range
 * 32.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 33 MHz - C_CTL_LPFH_RBB step size out of range
 * 33.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 34 MHz - C_CTL_LPFH_RBB step size out of range
 * 34.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 35 MHz - C_CTL_LPFH_RBB step size out of range
 * 35.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 36 MHz - C_CTL_LPFH_RBB step size out of range
 * 36.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 37 MHz - C_CTL_LPFH_RBB step size out of range
 * 37.5 MHz - C_CTL_LPFH_RBB step size out of range
 * 44 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 44.5 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 45 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 45.5 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 46 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 46.5 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 47 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 47.5 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)
 * 48 MHz - TuneVCO(CGEN) - failed to lock (cmphl != 2)

TX direction
====

``python SoapyLMS7/test_sweep_bandwidths.py --step=0.5e6 --start=1e6 --stop=60e6 --chan=0 --dir=TX``

 * 1 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 1.5 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 2 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 2.5 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 3 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 3.5 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 4 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 4.5 MHz - Tx lpf out of range 2.5-16 MHz and 25-65 MHz
 * 5 MHz - RCAL_LPFS5_TBB step size out of range
 * 5.5 MHz - RCAL_LPFS5_TBB step size out of range
 * 6 MHz - RCAL_LPFS5_TBB step size out of range

guruofquality, 2016-05-25:
Just to add to this, here is a user reported random failure setting the Rx bandwidth: https://gist.github.com/docprofsky/36daeba0e8f45a21263727210553c530#file-limesdr-soapy-probe-txt

guruofquality, 2016-06-04:
Was this a bug or simply a result of out of specification tuning? Should we merge this: https://github.com/myriadrf/LimeSuite/commit/e6c2c7fd64082d12e62be75e35e5b6dde673266e

rjonaitis, 2016-06-04:
The RF and IF frequencies were mixed up and some searches had value overflows. I merged in the SweepBandwidths branch.

### Rx

`python test_sweep_bandwidths.py --step=0.5e6 --start=1.5e6 --stop=130e6 --chan=0 --dir=RX`

For some reason these frequencies fail when used from the python script, but passes when used from LimeSuite GUI, needs further investigation.
 * 1.5 MHz - CFB_TIA_RFE step size out of range
 * 2 MHz - CFB_TIA_RFE step size out of range
 * 2.5 MHz - CFB_TIA_RFE step size out of range
 * 3 MHz - CFB_TIA_RFE step size out of range
 * 3.5 MHz - CFB_TIA_RFE step size out of range
 * 36 MHz - C_CTL_LPFH_RBB step size out of range

### Tx

`python test_sweep_bandwidths.py --step=0.5e6 --start=5e6 --stop=130e6 --chan=0 --dir=TX`

All tests passed

vsergeev, 2016-10-09:
+1, I'm seeing these errors when setting receive bandwidths lower than 4 MHz. It would be nice to access these bandwidths for oversampling or for lower sample rates.

```
[ERROR] setBandwidth(Rx, 0, 3 MHz) Failed - CFB_TIA_RFE step size out of range
```

romeojulietthotel, 2017-03-01:
Seeing these running Gnuradio.  Causes the block to fail.

No idea if this is significant but in Gnuradio terminal escape sequences are printed to the Gnuradio output window. Those sequences are only meant for a term window. But maybe that's irrelevant here and is a Gnuradio issue.

guruofquality, 2017-03-01:
> Seeing these running Gnuradio. Causes the block to fail.

What bandwidth was used when this happened?

> Also (but do not know if this is significant) in Gnuradio terminal escape sequences are printed to the Gnuradio output window. Those sequences are only meant for a term window. But maybe that's irrelevant here and is a Gnuradio issue.

SoapySDR by default prints the logger to stdout with escape sequences (for unix). So that wont come up right when GRC reads it and prints it in a gui text output. I think this is reason for the logging to be redirected with a custom handler to the gr logging facilities inside the gr-osmosdr soapy blocks.

romeojulietthotel, 2017-03-01:
Is this what we're looking for? It's possible that I am doing something wrong and I have tried different things. I don't want to pollute this issue but this seems to be the only place to mention this error.

[1m[31m[ERROR] setBandwidth(Rx, 0, 1.5 MHz) Failed - CFB_TIA_RFE step size out of range[0m

Will look at gr logging, didn't know about that, pardon the noise.

romeojulietthotel, 2017-03-02:
The error is not showing up.  I started a new grc block and no errors now.  I also realized that for my osmocom source I had not specified an antenna.   

I also do see a warning from libusb: `libusb: warning [libusb_exit] application left some devices open` and I am using libusbv1.0.20 and there is a .21 release now.

romeojulietthotel, 2017-03-04:
Nothing new to say except the problem persists. I am using latest git for all components.
This is all I did to trigger it. https://github.com/romeojulietthotel/flografs/blob/master/adsb.grc

FFY00, 2017-10-22:
You need to update the udev rules. Just do `./udev-rules/install.sh` from the LimeSuite repository root.

IgnasJarusevicius, 2018-06-22:
Filter calibrations has changed a lot since this issue was opened. I think it is fixed.

closed 2018-06-22 by IgnasJarusevicius