myriadrf/LimeSuite #61: LimeSDR OSX usb streaming issues

guruofquality, 2016-05-25:
Take a look at the errors and replicate on OSX system:
```
libusb: warning [darwin_abort_transfers] aborting all transactions on interface 0 pipe 2
libusb: error [libusb_cancel_transfer] cancel transfer failed error -99
```

Full dump from osmocom_fft: https://gist.github.com/docprofsky/ae873e525dac3e24475636df8fa2834d

tonycolo, 2017-07-13:
I just built LimeSuite from the master on Mac OS 10.12.5.
When I run the basicRX and simpleRX I see these errors sporadically, and when I shut down these programs, there is a significant number of them.  I've used LimeUtil --update to make sure the board's firmware is up to date.

IgnasJarusevicius, 2018-06-22:
Closing old issue. Re-open if still relevant.

closed 2018-06-22 by IgnasJarusevicius