myriadrf/LimeSuite #63: uLimeSDR - unresolved external symbol - MSVC 2015 x64

guruofquality, 2016-06-04:
I made a fix for x64: https://github.com/myriadrf/LimeSuite/commit/7da4fcd0150434e2ce55510aaa06d9bf2ea2cabd -- but its possible that the static libraries for FTD3XX may need to be available per version of MSVC:

```
  FTD3XX.lib(devcon.obj) : error LNK2019: unresolved external symbol __imp___iob_func referenced in function "void __cdecl Padding(int)" (?Padding@@Y
AXH@Z) [B:\LimeSuite\src\LimeUtil.vcxproj]
  B:\LimeSuite\src\Release\LimeUtil.exe : fatal error LNK1120: 1 unresolved externals [B:\LimeSuite\src\LimeUtil.vcxproj]
```

IgnasJarusevicius, 2016-06-05:
What problems does that x64 fix solve? Because now It no longer compiles using MSVC 2013. On machines that I tried CMake generates project that builds 32-bit executable and now it tries to link 64-bit library (because 64-bit CPU and OS), which of course fails. However, it is possible to succesfully build 64-bit executable by modifying MSVC project setting after running CMake (set platform to x64, link 64-bit WxWidgets and Cypress libraries). Does CMake generate project that (by default) targets x64 platform on your machine?

I tried building with MSVC 2015 and I get the same error as you when using static FTD3XX library. Linking against dynamic library seems to work fine. FTD3XX library is the most recent version and we don't have access to its source. WxWidgets also has problems with MSVC 2015 so I would sugest to use MSVC 2013 for now.

guruofquality, 2016-06-05:
> What problems does that x64 fix solve? Because now It no longer compiles using MSVC 2013. On machines that I tried CMake generates project that builds 32-bit executable and now it tries to link 64-bit library (because 64-bit CPU and OS), which of course fails.

My bad. Its supposed to detect the target architecture not the current CPU architecture. The check for 64 bit needs to be changed. But checking for 64 vs 32 and choosing the correct file is the right idea.

> I tried building with MSVC 2015 and I get the same error as you when using static FTD3XX library. Linking against dynamic library seems to work fine. FTD3XX library is the most recent version and we don't have access to its source.

AT least you see it as well. FTD3XX doesnt expose any C++ symbols as far as I can tell, I would expect the library to work on both compilers. Oh well, But I can try making an option for dynamic then. Static mode will be default, dynamic will install the FTD3XX dll.

> WxWidgets also has problems with MSVC 2015 so I would sugest to use MSVC 2013 for now.

FWIW, I have been building WX (latest 3.10 I believe) on both compilers with static libs. Maybe we can compare notes.

guruofquality, 2016-06-05:
@IgnasJarusevicius Hopefully this commit addresses all of the issues mentioned: https://github.com/myriadrf/LimeSuite/commit/227f6db2b367e5e1d59a453b98d711d2dcc6a4a8

We still have the static linking issue, but building dynamic for MSVC 2015 suits me needs anyway.

guruofquality, 2016-06-05:
I tested on MSVC2013 32-bit and MSVC2015 64-bit build and both build successfully. So I am closing this for now since we cant actually fix MSVC2015, but dynamic link is a good option. I will be able to enable uLimeSDR support when bundling limesuite in [PothosSDR installer](https://github.com/pothosware/PothosSDR/wiki) for 64-bit installers using both MSVC 2013 and 2015. Please reopen if the issue comes up again.

closed 2016-06-05 by guruofquality