myriadrf/LimeSuite #64: LimeSDR - 'L' during Tx

chemeris, 2016-06-05:
I'm seeing a lot of 'L' output in the console when transmitting using LimeSDR and UHD/Soapy, even at very small sample rate (1,083MSPS).
Is that a known issue?

guruofquality, 2016-06-05:
@chemeris I found a possible cause, description in commit: https://github.com/myriadrf/LimeSuite/commit/9c24803b5edc2908d91356602c5f0b72dca79189

And another fix that was surely causing problems: https://github.com/myriadrf/LimeSuite/commit/dc9300d8737365b6ea38eb27b5271407079c39c0

chemeris, 2016-06-06:
Great! Is this already fixed in Soapy/UHD?

guruofquality, 2016-06-06:
> Great! Is this already fixed in Soapy/UHD?

Should have just been an internal driver issue. If you recompile limesuite from master, i think it will be fine regardless of the upper layers.

chemeris, 2016-06-06:
Is Soapy using code from the LimeSuite? That's a surprising code layout.

I've been using pre-built binaries from the myriad ppa. Is there a timeline for fixing it there?

guruofquality, 2016-06-06:
> Is Soapy using code from the LimeSuite? That's a surprising code layout.

Actually having SoapySDR modules shipped along-side the driver or in place of any driver library was an original goal for SoapySDR. It just means that LimeSuite depends on SoapySDR

> I've been using pre-built binaries from the myriad ppa. Is there a timeline for fixing it there?

I can re-upload, but fixes are happening quite fast. Can you build LimeSuite from source, and use everything else from the PPA? Should be fine that way.

chemeris, 2016-06-06:
Ok, will build LimeSuite from master.

But in confused how rebuilding LimeSuite could help when I observe this when running osmo-trx with UHD/Soapy. Send like LimeSuite has nothing to do with this setup?

guruofquality, 2016-06-06:
LimeSuite has the entire implementation for LMS7002M, the LimeSDR usb driver, and the SoapyLMS7 wrapper around both. 

* FYI: apt-get instructions when building LimeSuite -- http://wiki.myriadrf.org/Lime_Suite#Ubuntu

chemeris, 2016-06-15:
Still getting L's with the Soapy LMS7 driver built from today's master. The only difference is that now I don't see debug output with bandwidth measurements. Any idea were to look next?

guruofquality, 2016-06-15:
> Still getting L's with the Soapy LMS7 driver built from today's master

So my fix addressed an issue when the time flag was inserted when it should not have been. This caused late packets to be seen when they should not have been. I think then -- in this case of osmotrx that the late packets are actually happening.

My suggestion is to print the hardware times when the late packets are indicated. If its misusing time, we should be able to catch it in the act.

And if not just prints in osmotrx, here is where the limesdr driver calls into the USB threaded endpoints, so prints there as well:

* write stream implementation will have the timestamp: https://github.com/myriadrf/LimeSuite/blob/master/src/ConnectionSTREAM/ConnectionSTREAMing.cpp#L460

* Call to get the current timestamp: https://github.com/myriadrf/LimeSuite/blob/master/src/ConnectionSTREAM/ConnectionSTREAMing.cpp#L609

* Where issue stream command timestamp goes: https://github.com/myriadrf/LimeSuite/blob/master/src/ConnectionSTREAM/ConnectionSTREAMing.cpp#L371

> don't see debug output with bandwidth measurements

I think they were removed, probably too much verbose. Its not a bug though.

chemeris, 2016-06-17:
Great, thank you. Tere were indeed later packets there. It seems LMS7 driver behavior is different from other UHD drivers like UmTRX and USRPs with regard to timestamp handing.

guruofquality, 2016-06-17:
Cool, closing then. I think if there is a suggestion for different policies to handle late timestamps, flushing, bursts, etc... lets make a new issue.

closed 2016-06-17 by guruofquality