myriadrf/LimeSuite #69: Rename lms7suite to LimeSuiteGUI

guruofquality, 2016-10-03:
The name for the gui executable has been lms7suite since before the rename and reorg. Thinking about a more limesuite-esque name. Perhaps "LimeSuiteGUI" would be more appropriate. Just wanted to bring up a discussion. Thoughts?

SimonG4ELI, 2016-10-04:
Just call it LMS Suite?

 

Simon Brown, GK4ELI
http://sdr-radio.com

9600, 2016-10-04:
@SimonG4ELI LMS Suite is the name for the legacy software that came in various versions for use with different boards:

https://github.com/myriadrf/lms-suite

For the history see:

https://myriadrf.org/blog/lms-suite-driver-discussion/
https://myriadrf.org/blog/limesuite-driver-architecture/

lms7suite is the legacy name for the GUI executable, so we need a new name for this which is distinct from the suite of software it forms part of and which reflects the fact it will support difference devices. Hence the suggestion of LimeSuiteGUI.

SimonG4ELI, 2016-10-04:
OK,

 

I was just think that there’s no real need to put GUI in the title.

 

Why not Lime Suite? Lime Console? 

 

Simon Brown, GK4ELI
http://sdr-radio.com

 

https://github.com/myriadrf/lms-suite

https://myriadrf.org/blog/lms-suite-driver-discussion/
https://myriadrf.org/blog/limesuite-driver-architecture/

9600, 2016-10-04:
@SimonG4ELI Lime Suite is a collection of software and not just a GUI; it would be like typing "linux" at the command prompt when you want to bring up a text editor.

guruofquality, 2016-10-05:
Thanks for the discussion, I wanted to take care of a rename before additional packaging work. The change is in master

closed 2016-10-10 by guruofquality