myriadrf/LimeSuite #72: SoapyLMS7 initialization takes around 1.5 seconds

vsergeev, 2016-10-09:
The SoapyLMS7 constructor takes about 1.2 seconds to open and initialize the LimeSDR, most of which is on [initializing default values](https://github.com/myriadrf/LimeSuite/blob/008db1b4b771e50d0ab557bd86dc416a190478f1/SoapyLMS7/Settings.cpp#L124-L145) of both channels in both directions (0.760 seconds).

The first call to `setSampleRate()` to set the application's sample rate then takes about 0.200 seconds: 0.160 seconds on `rfic->SetInterfaceFrequency()` and 0.040 seconds on `_conn->UpdateExternalDataRate()`. This totals to about 1.5 seconds to go from 
zero to samples on one channel. I haven't timed the internal functions yet, so I'm not sure where the bottlenecks are, but this does seem a little bit long and strange. Speeding this up, if possible, would benefit prototyping uses that have frequent restarts.

guruofquality, 2017-01-10:
Update: Fixed with the latest software and fw/gw images using faster FPGA SPI implementation. :-)

closed 2017-01-10 by guruofquality

vsergeev, 2017-07-31:
Awesome, thanks.