myriadrf/LimeSuite #74: No samples at ~29 to 30 MHz sample rate

vsergeev, 2016-10-09:
At sample rates of 29 to 30 MHz, the LimeSDR LEDS1 (furthest from the USB3 port and closest to board edge) flashes red and no samples are streamed over USB. Sample rates below or above this range work great.

Short test program to reproduce:

```
#include <SoapySDR/Device.h>
#include <SoapySDR/Formats.h>
#include <stdio.h>
#include <stdlib.h>
#include <complex.h>
#include <assert.h>

int main(void) {
    SoapySDRDevice *sdr;
    SoapySDRStream *rxStream;

    SoapySDRKwargs args = {};
    SoapySDRKwargs_set(&args, "driver", "lime");
    assert((sdr = SoapySDRDevice_make(&args)) != NULL);
    SoapySDRKwargs_clear(&args);

    assert(SoapySDRDevice_setSampleRate(sdr, SOAPY_SDR_RX, 0, 29e6) == 0);
    assert(SoapySDRDevice_setFrequency(sdr, SOAPY_SDR_RX, 0, 88.5e6, NULL) == 0);
    assert(SoapySDRDevice_setupStream(sdr, &rxStream, SOAPY_SDR_RX, SOAPY_SDR_CF32, (size_t [1]){0}, 1, NULL) == 0);
    assert(SoapySDRDevice_activateStream(sdr, rxStream, 0, 0, 0) == 0);

    complex float buff[10*SoapySDRDevice_getStreamMTU(sdr, rxStream)];
    int flags; long long timeNs;
    while (1) {
        void *buffs[] = {buff};
        int ret = SoapySDRDevice_readStream(sdr, rxStream, buffs, sizeof(buff)/sizeof(complex float), &flags, &timeNs, 1e12);
        printf("%d\n", ret);
        assert(ret > 0);
    }

    assert(SoapySDRDevice_deactivateStream(sdr, rxStream, 0, 0) == 0);
    assert(SoapySDRDevice_closeStream(sdr, rxStream) == 0);
    SoapySDRDevice_unmake(sdr);

    return 0;
}
```

Output, at 29 MHz sample rate:

```
$ ./test
[INFO] Make connection: 'USB 3.0 (LimeSDR-USB)'
CLK0 fOut = 27 MHz  Multisynth Divider 33 0/1  R divider = 1 source = PLLA
CLK1 fOut = 27 MHz  Multisynth Divider 33 0/1  R divider = 1 source = PLLA
CLK2 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
CLK3 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
CLK4 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
CLK5 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
CLK6 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
CLK7 fOut = 27 MHz  Multisynth Divider 8 0/1  R divider = 1 source = PLLA
Si5351C : VCOA = 891 MHz  Feedback Divider 35 41943/65536

Si5351C : VCOB = 891 MHz  Feedback Divider 35 41943/65536

[INFO] Device name: LimeSDR USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
LMS7002M values cache at /home/anteater/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=0
[INFO] LMS7002M calibration values caching Enable
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 0, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 0, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 0, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 0, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
[INFO] SoapyLMS7::setSampleRate(Tx, 0, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
[INFO] SoapyLMS7::setBandwidth(Rx, 0, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 0, 30 MHz)
[INFO] Tx Filter calibrated from cache
[INFO] SoapyLMS7::setFrequency(Rx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 1, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 1, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 1, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 1, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 1, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
[INFO] SoapyLMS7::setSampleRate(Tx, 1, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
[INFO] SoapyLMS7::setBandwidth(Rx, 1, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 1, 30 MHz)
[INFO] Tx Filter calibrated from cache
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 29 MHz), CGEN=464 MHz, ADC=116 MHz, decim=4
ConnectionSTREAM::ConfigureFPGA_PLL(tx=5MHz, rx=5MHz)
[INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 88.5 MHz)
SetFrequency using cache values vco:1, csw:113
[WARNING] GetDC_IQ_Interp(88.5 MHz, ch=0, tx=0): no matches between [87.5, 89.5] MHz
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)

```

Maybe the dividers are computed incorrectly when setting a sample rate close to the reference frequency?

IgnasJarusevicius, 2016-10-27:
Soapy wrapper modified to avoid frequencies where CGEN cannot lock.
7e01c54a6ec8ac5668a9f497d231896ee5df3e69

closed 2016-10-27 by IgnasJarusevicius

vsergeev, 2016-11-07:
:+1:, building with tip 28275121f07e3a4726e8191cc399e688d31c02a6, it appears to stream samples now at 29-30 MHz and 14.5 - 15 MHz ranges. Thanks.