myriadrf/LimeSuite #76: Infinite data from readStreamStatus()

guruofquality, 2016-10-18:
At least from the ``int SoapyLMS7::readStreamStatus(`` side of things, this is a call thats meant to block until there is an event like an underflow, late packet, etc. There is code that loops on this API call in various programs (example SoapyRemote), and that creates a lot of data or prints when it can return ASAP with return code 0 every time.

The issue comes up with the way ``int ILimeSDRStreaming::ReadStreamStatus(`` is implemented. My thoughts are to give it a condition variable which we can notify when a stream event happens. Either way there is something to change. Just leaving this issue open for some discussion.

rjonaitis, 2016-10-20:
Having `int ILimeSDRStreaming::ReadStreamStatus` as a blocking function would require host application to explicitly create a new thread to use it. It would prevent usage of LimeSuite library in other software which are limited to one thread, such as Matlab or Octave. It would be better to make a callback for notification about underflow, late packets, etc. That callback then could be used inside Soapy wrapper to notify the conditional variable for  `int SoapyLMS7::readStreamStatus`

guruofquality, 2016-10-24:
> Having int ILimeSDRStreaming::ReadStreamStatus as a blocking function would require host application to explicitly create a new thread to use it. It would prevent usage of LimeSuite library in other software which are limited to one thread, such as Matlab or Octave.

At present ReadStreamStatus() has a timeout_ms parameter. So it would be possible to implement a polling API with timeout_ms = 0, and an event blocking API when timeout is used. Just a thought. If the call is never intended to block, perhaps the timeout should be removed.

> It would be better to make a callback for notification about underflow, late packets, etc. That callback then could be used inside Soapy wrapper to notify the conditional variable for int SoapyLMS7::readStreamStatus

I do like the callback idea for various "events" that happen within the stream threads

closed 2018-06-22 by IgnasJarusevicius