myriadrf/LimeSuite #81: LMS_SetLOFrequency not work when freq<30mhz

jocover, 2017-01-02:
could add Automatically HF mode feature on LMS API like #62

jocover, 2017-01-04:
my limesdr hardware versione 1.2 os win10
my test samplerate 8mhz and freq 20mhz
I found setting LMS_SetLOFrequency freq lower than 30Mhz after LMS_StartStream, this is not work

this is my mod basicRX.cpp test code
> auto t1 = chrono::high_resolution_clock::now();
**size_t count = 0;**
while ( chrono::high_resolution_clock::now() - t1 < chrono::seconds( **20** ) ) /* run for 5 seconds */
{
	int samplesRead;
	/* Receive samples */
	samplesRead = LMS_RecvStream( &streamId, buffer, bufersize, NULL, 1000 );
	/* I and Q samples are interleaved in buffer: IQIQIQ... */
	**if ( count % 10 == 0 )
	{
	    printf( "Count:%d Received %d samples\n", count, samplesRead );
	}
	count++;
	if ( count == 200 )
	{
	    LMS_SetLOFrequency( device, LMS_CH_RX, 0, 20e6 );
	    printf( "setting freq\n" );
	}**

or using my test version sdrsharp plugin
https://mega.nz/#!3RElmDxD!_YqGHtKOHc8Y-LRSb1FUEUjz0v7sjsUnjokmOUuuwDQ

IgnasJarusevicius, 2017-09-22:
Closing. Old issue. 
LMS_SetLOFrequency() supports setting frequency to <30 MHz.
Also LimeSDR v1.2 are no longer supported so if there are problems with v1.2 boards they will not be resolved.

closed 2017-09-22 by IgnasJarusevicius