myriadrf/LimeSuite #82: Compilcation error: ConnectionNovenaRF7Entry.cpp

gustad, 2017-01-10:
```
git clone https://github.com/myriadrf/LimeSuite.git LimeSuite.git
cd LimeSuite.git/
mkdir builddir && cd builddir
cmake ../
git rev-parse --short HEAD
54dadad
gcc --version |head -1
gcc (Gentoo 4.9.4 p1.0, pie-0.6.4) 4.9.4
uname -r
4.4.26-gentoo
make

```
results in

```
/home/projects/LimeSuite.git/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp: In function ‘bool IsNovenaBoard()’:
/home/projects/LimeSuite.git/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp:36:30: error: elements of array ‘i2c_msg messages [2]’ have incomplete type
     struct i2c_msg messages[2];

```

There is only one instance in the repository:

```
git grep i2c_msg | cat
src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp:    struct i2c_msg messages[2];
```
Then I did:

```
git diff |cat
diff --git a/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp b/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp
index 207a7af..643e309 100644
--- a/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp
+++ b/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp
@@ -18,6 +18,7 @@
 #include <linux/types.h>
 #include <linux/spi/spidev.h>
 #include <linux/i2c-dev.h>
+#include <linux/i2c.h>
 #include <iostream>
```
Now I get:

```
 make
[  1%] Building CXX object src/CMakeFiles/LimeSuite.dir/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp.o
/home/projects/LimeSuite.git/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp: In function ‘bool IsNovenaBoard()’:
/home/projects/LimeSuite.git/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp:46:21: error: invalid conversion from ‘char*’ to ‘__u8* {aka unsigned char*}’ [-fpermissive]
     messages[0].buf = set_addr_buf;
                     ^
/home/projects/LimeSuite.git/src/ConnectionNovenaRF7/ConnectionNovenaRF7Entry.cpp:50:21: error: invalid conversion from ‘char*’ to ‘__u8* {aka unsigned char*}’ [-fpermissive]
     messages[1].buf = data;
```

Is a specific version of gcc required?

IgnasJarusevicius, 2017-01-10:
Do you have libi2c-dev installed? To install it:
> sudo apt-get install libi2c-dev

Also I don't know which board you are using, but if it is not Novena-RF you can safely disable it in CMake ( 'ENABLE_NOVENARF7' option in CMakeCache.txt )

guruofquality, 2017-01-10:
This comes up a lot, so I added a check for the userspace version of libi2c headers: https://github.com/myriadrf/LimeSuite/commit/8629033b56bf4fc04ab8192d6d8271c530231904 Its in stable now, will merge into master.

closed 2017-01-10 by guruofquality

gustad, 2017-01-10:
Thanks,

I don't have a board yet. I'm waiting for my LimeSDR to arrive and in the meantime I'm trying to set up my environment to be able to build FPGA, FX3 firmare, and LimeSuite.

But isn't the i2c_msg stuct which is a part of the Linux kernel the correct one to use?

The i2c_msg struct is defined include/linux/i2c.h (which again includes include/uapi/linux/i2c.h).

If I include i2c.h my source will compile just fine if I add some casts:

https://gist.github.com/gustad/dbb42de8a53586157fe63a1bbde22f8c

Or does LimeSuite require a total different i2c_msg struct than the one in the kernel?

guruofquality, 2017-01-10:
The package libi2c-dev (I'm not sure how pervasive this package is outside of debian based systems) provides a different version of i2c.h that uses standard C types and it might have some other typedefs or structs as well. But I'm pretty sure its ABI compatible with the equivalent linux header.

Anyway, the novena rf support uses this version of the header and not the kernel version. Also I should note that the novena support only needs to run on the novena board itself, which is debian based. And so there isnt anything missing to not compile novena support on a standard linux PC.

gustad, 2017-01-10:
Thank you for the explanation.

Looking at:
https://packages.debian.org/jessie/all/libi2c-dev/filelist

Seems like it will overwrite the kernel source file in /usr/include/linux/i2c-dev.h. 

I can't seem to find anything like this on Gentoo. Could be a bit scary if the Linux kernel struct layout should change at some point in time. Anyways, like you say I will do without on the LimeSDR when I get it...