myriadrf/LimeSuite #86: Setup travis and appveyor CI

guruofquality, 2017-01-14:
Automatic compilation checks and unit tests and reporting any failure. Also good for confirming that pull requests do not break compile before merging.

* [ ] Setup .travis.yaml for maser/stable and PRs
* [ ] Allow github app access to travis CI
* [ ] Setup appveyor.yaml for maser/stable and PRs
* [ ] Allow github app access to appveyor CI

Reference:
* For linux: https://travis-ci.org/
* For windows: https://www.appveyor.com/

guruofquality, 2017-01-16:
@jocover thats awesome, thanks!

@9600 can you enable these services when you get a chance? Basically you can login with github account that has admin rights and enable which repos get CI access.

9600, 2017-01-16:
@guruofquality enabled them both for this repo and if we need it for any others drop me a line with details. Also taken default options for both, so let me know if anything needs changing.

guruofquality, 2017-01-16:
@9600 so far so good. Can you login to appveyor and share the link under the settings -> badges there is a raster image url that looks like https://ci.appveyor.com/api/projects/status/some_unique_id_here

I'm just looking for that URL for a status badge, but I couldn’t find the ID otherwise from here: https://ci.appveyor.com/project/9600/limesuite

9600, 2017-01-16:
Project status: https://ci.appveyor.com/api/projects/status/wjmtvygjg67tnhrf

Master branch status: https://ci.appveyor.com/api/projects/status/wjmtvygjg67tnhrf/branch/master

However, looks like I should probably create a new user and set this up as an organisation which is mapped onto our GitHub one. If I do this, I'm assuming we just need to remove the project from under my account and then re-add it under the organisation?

guruofquality, 2017-01-16:
I see. When I setup appveyor I also used my account. But it looks like its possible to do this as an organization: http://help.appveyor.com/discussions/questions/1154-appveyor-account-for-github-organizations It should be safe to just remove and re-add it like you said. Its possible that the url mentioned above will change though.

9600, 2017-01-16:
Deleted the project from my Appveyor account and recreated it under a new "myriadrf" one:

https://ci.appveyor.com/project/myriadrf/limesuite

Project status: https://ci.appveyor.com/api/projects/status/stwfarvq0j01qyaq

Master branch status: https://ci.appveyor.com/api/projects/status/stwfarvq0j01qyaq/branch/master

guruofquality, 2017-01-16:
Looks like I will move this to the myriadrf downloads: `appveyor DownloadFile https://github.com/jocover/LimeSuite/releases/download/fx3sdk/FX3SDK.zip FX3SDK.zip`

guruofquality, 2017-01-18:
Thanks everybody!

closed 2017-01-18 by guruofquality