myriadrf/LimeSuite #90: GUI not fitting screen 1366x768

c5e3, 2017-01-20:
I have another layout issue; at 1366x768 the window is cut off at the tab 'LDO'.
I think, a simple scrollbar would help. The windows is not scalable, so i have no chance of seeing the other checkboxes

version: 16.12.0-myriadrf1~yakkety on ubuntu 16.10 4.8.0-34-generic x86_64

[image]

romeojulietthotel, 2017-02-01:
Seeing this as well. Built from git master and wxwidget git master.
Commit  [ce9f6ef7ccc0](https://github.com/myriadrf/LimeSuite/commit/ce9f6ef7ccc0765cfbabc0656472d0d3bd23a88c#diff-eafd592852631f81815b069b06c335f0) seems to be the bad commit.

romeojulietthotel, 2017-02-08:
I wanted to test the waters to see if you would want a pull request. In the current state it is not really usable. As @c5e3 has noted the window is not resizable so there's really no relief.

I don't know wxwidgets or limesuite but I can look at it.  But maybe someone is already working on this.

romeojulietthotel, 2017-02-10:
I looked at wxFormBuilder (which seems to be a requirement) but wxFormBuilder will not compile against wxWidgets >= 3.1 so I cannot try to work on this.

IgnasJarusevicius, 2017-09-22:
Closing. I think recent GUI versions fit on 1366x768 screen. Proper window size control is not implemented but that seem same as issue #110

closed 2017-09-22 by IgnasJarusevicius

romeojulietthotel, 2017-09-22:
> I think recent GUI versions fit on 1366x768 screen.

**Hello** @IgnasJarusevicius 
Unfortunately they don't fit. I just tried with a fresh build of git master on an updated system.