myriadrf/LimeSuite #93: new / delete[] mismatch

mpbraendli, 2017-02-10:
Hello,
running ```LimeUtil --make``` through valgrind found a few issues.

```ConnectionNovenaRF7::StreamChannel::Write()``` needs to ```delete[] samplesShort``` (it uses ```delete```) instead.

Same in ```ConnectionSTREAM::Write()``` for ```wbuffer```, ```ILimeSDRStreaming::StreamChannel::Write()``` for ```samplesShort``` and ```LMS64CProtocol::TransferPacket()``` for ```outBuffer```, because they are all allocated with ```new[]``` and not ```new```.

Related question: would you consider pull requests that replace such manual buffer management by ```std::vector<uint8_t>```? (As long as it's internal and doesn't impact the API)

IgnasJarusevicius, 2017-02-13:
Hello,
Thanks for spotting this. I have fixed it with d94e2b8d04c8a3acf0eee4a2df74a7fb8b8a9f76

As for replacement with std::vector, I personally would prefer that it stays as it is.

mpbraendli, 2017-02-13:
Thank you.
Ok for keeping the buffer management manual.

closed 2017-02-13 by mpbraendli