myriadrf/LimeSuite #96: undefined reference to libusb

KC7NOA, 2017-02-24:
[ 87%] Building CXX object src/CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlLimeSDR.cpp.o
[ 87%] Linking CXX executable ../bin/LimeSuiteGUI
libLimeSuite.so.17.02.0: undefined reference to libusb_control_transfer'
libLimeSuite.so.17.02.0: undefined reference tolibusb_get_device_descriptor'
libLimeSuite.so.17.02.0: undefined reference to libusb_strerror'
libLimeSuite.so.17.02.0: undefined reference tolibusb_set_debug'
libLimeSuite.so.17.02.0: undefined reference to libusb_handle_events_timeout_completed'
libLimeSuite.so.17.02.0: undefined reference tolibusb_open_device_with_vid_pid'
libLimeSuite.so.17.02.0: undefined reference to libusb_submit_transfer'
libLimeSuite.so.17.02.0: undefined reference tolibusb_close'
libLimeSuite.so.17.02.0: undefined reference to libusb_init'
libLimeSuite.so.17.02.0: undefined reference tolibusb_free_transfer'
libLimeSuite.so.17.02.0: undefined reference to libusb_detach_kernel_driver'
libLimeSuite.so.17.02.0: undefined reference tolibusb_open'
libLimeSuite.so.17.02.0: undefined reference to libusb_get_device_speed'
libLimeSuite.so.17.02.0: undefined reference tolibusb_cancel_transfer'
libLimeSuite.so.17.02.0: undefined reference to libusb_release_interface'
libLimeSuite.so.17.02.0: undefined reference tolibusb_get_device_list'
libLimeSuite.so.17.02.0: undefined reference to libusb_exit'
libLimeSuite.so.17.02.0: undefined reference tolibusb_get_string_descriptor_ascii'
libLimeSuite.so.17.02.0: undefined reference to libusb_get_device'
libLimeSuite.so.17.02.0: undefined reference tolibusb_kernel_driver_active'
libLimeSuite.so.17.02.0: undefined reference to libusb_get_active_config_descriptor'
libLimeSuite.so.17.02.0: undefined reference tolibusb_bulk_transfer'
libLimeSuite.so.17.02.0: undefined reference to libusb_alloc_transfer'
libLimeSuite.so.17.02.0: undefined reference tolibusb_free_device_list'
libLimeSuite.so.17.02.0: undefined reference to libusb_error_name'
libLimeSuite.so.17.02.0: undefined reference tolibusb_reset_device'
libLimeSuite.so.17.02.0: undefined reference to libusb_claim_interface'
libLimeSuite.so.17.02.0: undefined reference tolibusb_free_config_descriptor'
collect2: error: ld returned 1 exit status
src/CMakeFiles/LimeSuiteGUI.dir/build.make:1528: recipe for target 'bin/LimeSuiteGUI' failed
make[2]: *** [bin/LimeSuiteGUI] Error 1
CMakeFiles/Makefile2:172: recipe for target 'src/CMakeFiles/LimeSuiteGUI.dir/all' failed
make[1]: *** [src/CMakeFiles/LimeSuiteGUI.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

This might be my fault ... right now im not sure.
Arm Odroid-C2 Ubuntu 16.04

KC7NOA, 2017-02-24:
Only difference that i see between my Odroid-X2(32bit) and my Odroid-C2 (64bit) is that the C2 still has the warning that 

NovenaRF7 Connection support is missing/disabled

guruofquality, 2017-02-24:
Can you post the output of the cmake command? It may not have correctly found libusb1.0 devel files

KC7NOA, 2017-02-24:
Odroid-C2 64Bit ######

kc7noaa@odroid64:~/Downloads/LimeSuite/build$ cmake ../
-- SQLITE3: TRUE
-- SQLITE3_INCLUDE_DIRS: /usr/include
-- SQLITE3_LIBRARIES: /usr/lib/aarch64-linux-gnu/libsqlite3.so
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/aarch64-linux-gnu/libusb.so
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/aarch64-linux-gnu/libusb.so
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/aarch64-linux-gnu/libusb.so
-- wxWidgets_DEFINITIONS: _FILE_OFFSET_BITS=64;WXUSINGDLL;__WXGTK__
-- wxWidgets_INCLUDE_DIRS: /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0;/usr/include/wx-3.0
-- wxWidgets_LIBRARIES: -L/usr/lib/aarch64-linux-gnu;-pthread;;;-lwx_gtk2u_richtext-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_xrc-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0;-lwx_gtk2u_gl-3.0
-- GNUPlot Found
-- 
-- ######################################################
-- ## LimeSuite enabled features
-- ######################################################
-- 
 * LimeSuiteHeaders , The lime suite headers
 * LimeSuiteLibrary , The lime suite library
 * ConnectionSTREAM , STREAM Connection support
 * Connection_uLimeSDR , uLimeSDR Connection support
 * ConnectionXillybus , PCIE Xillybus Connection support
 * LimeSuiteGUI , GUI Application for LimeSuite
 * LimeUtilCommand , Command line device discovery utility
 * LimeSuiteExamples , LimeSuite library API examples
 * SoapySDRLMS7 , SoapySDR bindings for LMS7
 * LimeSuiteDesktop , LimeSuite freedesktop integration

-- ######################################################
-- ## LimeSuite disabled features
-- ######################################################
-- 
 * ConnectionEVB7COM , EVB+COM Connection support
 * ConnectionSTREAM_UNITE , STREAM+UNITE Connection support
 * ConnectionNovenaRF7 , NovenaRF7 Connection support
 * LimeSuiteUtilities , Additional utility programs for testing
 * LimeSuiteTests , Download gtest and build test application

-- Install prefix: /usr/local
-- Build timestamp: 2017-02-24
-- Lime Suite version: 17.02.0-g7a5b820d
-- ABI/so version: 17.02-1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kc7noaa/Downloads/LimeSuite/build
kc7noaa@odroid64:~/Downloads/LimeSuite/build$ 

Odroid-X2 32Bit ######

kc7noaodroid@odroid:~/Downloads/LimeSuite/build$ cmake ../
-- SQLITE3: TRUE
-- SQLITE3_INCLUDE_DIRS: /usr/include
-- SQLITE3_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libsqlite3.so
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- wxWidgets_DEFINITIONS: _FILE_OFFSET_BITS=64;WXUSINGDLL;__WXGTK__
-- wxWidgets_INCLUDE_DIRS: /usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0;/usr/include/wx-3.0
-- wxWidgets_LIBRARIES: -L/usr/lib/arm-linux-gnueabihf;-pthread;;;-lwx_gtk2u_richtext-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_xrc-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0;-lwx_gtk2u_gl-3.0
-- GNUPlot Found
-- 
-- ######################################################
-- ## LimeSuite enabled features
-- ######################################################
-- 
 * LimeSuiteHeaders , The lime suite headers
 * LimeSuiteLibrary , The lime suite library
 * ConnectionSTREAM , STREAM Connection support
 * ConnectionNovenaRF7 , NovenaRF7 Connection support
 * Connection_uLimeSDR , uLimeSDR Connection support
 * ConnectionXillybus , PCIE Xillybus Connection support
 * LimeSuiteGUI , GUI Application for LimeSuite
 * LimeUtilCommand , Command line device discovery utility
 * LimeSuiteExamples , LimeSuite library API examples
 * SoapySDRLMS7 , SoapySDR bindings for LMS7
 * LimeSuiteDesktop , LimeSuite freedesktop integration

-- ######################################################
-- ## LimeSuite disabled features
-- ######################################################
-- 
 * ConnectionEVB7COM , EVB+COM Connection support
 * ConnectionSTREAM_UNITE , STREAM+UNITE Connection support
 * LimeSuiteUtilities , Additional utility programs for testing
 * LimeSuiteTests , Download gtest and build test application

-- Install prefix: /usr/local
-- Build timestamp: 2017-02-24
-- Lime Suite version: 17.02.0-g7a5b820d
-- ABI/so version: 17.02-1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kc7noaodroid/Downloads/LimeSuite/build
kc7noaodroid@odroid:~/Downloads/LimeSuite/build$

KC7NOA, 2017-02-24:
Just making sure ...

kc7noaa@odroid64:~/Downloads/LimeSuite/build$ sudo apt-get install libusb-1.0-0-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version (2:1.0.20-1).
The following packages were automatically installed and are no longer required:
  libcec3 linux-tools-4.4.0-41 linux-tools-4.4.0-41-generic
  linux-tools-4.4.0-58 linux-tools-4.4.0-58-generic linux-tools-4.4.0-64
  linux-tools-4.4.0-64-generic ubuntu-core-launcher
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
kc7noaa@odroid64:~/Downloads/LimeSuite/build$

guruofquality, 2017-02-25:
Well it found libusb 1.0 so thats interesting. Can you post the output of make with verbose: `make VERBOSE=1`? That will show whats getting passed to the linker.

KC7NOA, 2017-02-25:
This should be plenty ... or do you want a paste of the error ...

kc7noaa@odroid64:~/Downloads/LimeSuite/build$ make VERBOSE=1
/usr/bin/cmake -H/home/kc7noaa/Downloads/LimeSuite -B/home/kc7noaa/Downloads/LimeSuite/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/kc7noaa/Downloads/LimeSuite/build/CMakeFiles /home/kc7noaa/Downloads/LimeSuite/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/kc7noaa/Downloads/LimeSuite/build'
make -f src/CMakeFiles/LimeSuite.dir/build.make src/CMakeFiles/LimeSuite.dir/depend
make[2]: Entering directory '/home/kc7noaa/Downloads/LimeSuite/build'
cd /home/kc7noaa/Downloads/LimeSuite/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kc7noaa/Downloads/LimeSuite /home/kc7noaa/Downloads/LimeSuite/src /home/kc7noaa/Downloads/LimeSuite/build /home/kc7noaa/Downloads/LimeSuite/build/src /home/kc7noaa/Downloads/LimeSuite/build/src/CMakeFiles/LimeSuite.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/kc7noaa/Downloads/LimeSuite/build'
make -f src/CMakeFiles/LimeSuite.dir/build.make src/CMakeFiles/LimeSuite.dir/build
make[2]: Entering directory '/home/kc7noaa/Downloads/LimeSuite/build'
[  0%] Building CXX object src/CMakeFiles/LimeSuite.dir/ErrorReporting.cpp.o
cd /home/kc7noaa/Downloads/LimeSuite/build/src && /usr/bin/c++   -DLIME_DLL -DLIME_DLL_EXPORTS -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -I/home/kc7noaa/Downloads/LimeSuite/src -I/home/kc7noaa/Downloads/LimeSuite/src/API -I/home/kc7noaa/Downloads/LimeSuite/src/GFIR -I/home/kc7noaa/Downloads/LimeSuite/src/protocols -I/home/kc7noaa/Downloads/LimeSuite/src/ConnectionRegistry -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002m_mcu -I/home/kc7noaa/Downloads/LimeSuite/src/ADF4002 -I/home/kc7noaa/Downloads/LimeSuite/src/Si5351C -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002m -I/home/kc7noaa/Downloads/LimeSuite/src/LTEpackets -I/home/kc7noaa/Downloads/LimeSuite/src/FPGA_common -I/home/kc7noaa/Downloads/LimeSuite/external/cpp-feather-ini-parser -I/home/kc7noaa/Downloads/LimeSuite/src/HPM7 -I/home/kc7noaa/Downloads/LimeSuite/src/kissFFT -I/usr/include/libusb-1.0 -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/LMS_Programing -I/home/kc7noaa/Downloads/LimeSuite/src/utilities_gui -I/home/kc7noaa/Downloads/LimeSuite/src/RFSpark -I/home/kc7noaa/Downloads/LimeSuite/src/FPGAcontrols_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/numericSlider -I/home/kc7noaa/Downloads/LimeSuite/src/myriad7 -I/home/kc7noaa/Downloads/LimeSuite/src/fftviewer_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/lms7suiteEvents -I/home/kc7noaa/Downloads/LimeSuite/src/boards_wxgui -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0  -fvisibility-inlines-hidden -pthread -O3 -DNDEBUG -fPIC   -Wall -fvisibility=hidden -std=gnu++11 -o CMakeFiles/LimeSuite.dir/ErrorReporting.cpp.o -c /home/kc7noaa/Downloads/LimeSuite/src/ErrorReporting.cpp
[  1%] Building CXX object src/CMakeFiles/LimeSuite.dir/ADF4002/ADF4002.cpp.o
cd /home/kc7noaa/Downloads/LimeSuite/build/src && /usr/bin/c++   -DLIME_DLL -DLIME_DLL_EXPORTS -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -I/home/kc7noaa/Downloads/LimeSuite/src -I/home/kc7noaa/Downloads/LimeSuite/src/API -I/home/kc7noaa/Downloads/LimeSuite/src/GFIR -I/home/kc7noaa/Downloads/LimeSuite/src/protocols -I/home/kc7noaa/Downloads/LimeSuite/src/ConnectionRegistry -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002m_mcu -I/home/kc7noaa/Downloads/LimeSuite/src/ADF4002 -I/home/kc7noaa/Downloads/LimeSuite/src/Si5351C -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002m -I/home/kc7noaa/Downloads/LimeSuite/src/LTEpackets -I/home/kc7noaa/Downloads/LimeSuite/src/FPGA_common -I/home/kc7noaa/Downloads/LimeSuite/external/cpp-feather-ini-parser -I/home/kc7noaa/Downloads/LimeSuite/src/HPM7 -I/home/kc7noaa/Downloads/LimeSuite/src/kissFFT -I/usr/include/libusb-1.0 -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/LMS_Programing -I/home/kc7noaa/Downloads/LimeSuite/src/utilities_gui -I/home/kc7noaa/Downloads/LimeSuite/src/RFSpark -I/home/kc7noaa/Downloads/LimeSuite/src/FPGAcontrols_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/numericSlider -I/home/kc7noaa/Downloads/LimeSuite/src/myriad7 -I/home/kc7noaa/Downloads/LimeSuite/src/fftviewer_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/lms7suiteEvents -I/home/kc7noaa/Downloads/LimeSuite/src/boards_wxgui -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0  -fvisibility-inlines-hidden -pthread -O3 -DNDEBUG -fPIC   -Wall -fvisibility=hidden -std=gnu++11 -o CMakeFiles/LimeSuite.dir/ADF4002/ADF4002.cpp.o -c /home/kc7noaa/Downloads/LimeSuite/src/ADF4002/ADF4002.cpp
[  2%] Building CXX object src/CMakeFiles/LimeSuite.dir/lms7002m_mcu/MCU_BD.cpp.o
cd /home/kc7noaa/Downloads/LimeSuite/build/src && /usr/bin/c++   -DLIME_DLL -DLIME_DLL_EXPORTS -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__WXGTK__ -I/home/kc7noaa/Downloads/LimeSuite/src -I/home/kc7noaa/Downloads/LimeSuite/src/API -I/home/kc7noaa/Downloads/LimeSuite/src/GFIR -I/home/kc7noaa/Downloads/LimeSuite/src/protocols -I/home/kc7noaa/Downloads/LimeSuite/src/ConnectionRegistry -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002m_mcu -I/home/kc7noaa/Downloads/LimeSuite/src/ADF4002 -I/home/kc7noaa/Downloads/LimeSuite/src/Si5351C -I/home/kc7noaa/Downloads/LimeSuite/src/lms7002m -I/home/kc7noaa/Dow [... cut ...] ad7 -I/home/kc7noaa/Downloads/LimeSuite/src/fftviewer_wxgui -I/home/kc7noaa/Downloads/LimeSuite/src/lms7suiteEvents -I/home/kc7noaa/Downloads/LimeSuite/src/boards_wxgui -isystem /usr/lib/aarch64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0  -fvisibility-inlines-hidden -pthread -O3 -DNDEBUG -fPIC   -Wall -fvisibility=hidden -std=gnu++11 -o CMakeFiles/LimeSuite.dir/ConnectionRegistry/IConnection.cpp.o -c /home/kc7noaa/Downloads/LimeSuite/src/ConnectionRegistry/IConnection.cpp

KC7NOA, 2017-02-25:
and the error ...

[ 81%] Linking CXX executable ../bin/LimeSuiteGUI
cd /home/kc7noaa/Downloads/LimeSuite/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/LimeSuiteGUI.dir/link.txt --verbose=1
/usr/bin/c++    -fvisibility-inlines-hidden -pthread -O3 -DNDEBUG   CMakeFiles/LimeSuiteGUI.dir/ADF4002/ADF4002_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlBuffers_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlXBUF_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlRFE_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlCalibrations_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlTRF_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlCDS_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlLimeLightPAD_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlBIST_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/dlgViewIRAM.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_gui_utilities.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlSX_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlTBB_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_dlgVCOfrequencies.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlRBB_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlLDO_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlAFE_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/CoefficientFileParser.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlRxTSP_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_mainPanel.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/pnlBoardControls.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/dlgViewSFR.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlCLKGEN_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlMCU_BD_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlBIAS_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_dlgGFIR_Coefficients.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlTxTSP_view.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7002_wxgui/lms7002_pnlR3.cpp.o CMakeFiles/LimeSuiteGUI.dir/LMS_Programing/LMS_Programing_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/utilities_gui/dlgConnectionSettings.cpp.o CMakeFiles/LimeSuiteGUI.dir/utilities_gui/pnlMiniLog.cpp.o CMakeFiles/LimeSuiteGUI.dir/utilities_gui/SPI_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/utilities_gui/dlgFullMessageLog.cpp.o CMakeFiles/LimeSuiteGUI.dir/utilities_gui/dlgDeviceInfo.cpp.o CMakeFiles/LimeSuiteGUI.dir/utilities_gui/utilities_gui.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7suiteAppFrame.cpp.o CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/lms7002m_novena_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/RFSpark/RFSpark_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/Si5351C/Si5351C_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7suiteApp_gui.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7suiteApp.cpp.o CMakeFiles/LimeSuiteGUI.dir/FPGAcontrols_wxgui/FPGAcontrols_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/HPM7/HPM7_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/dlgAbout.cpp.o CMakeFiles/LimeSuiteGUI.dir/numericSlider/numericSlider.cpp.o CMakeFiles/LimeSuiteGUI.dir/myriad7/myriad7_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/fftviewer_wxgui/fftviewer_wxgui.cpp.o CMakeFiles/LimeSuiteGUI.dir/fftviewer_wxgui/fftviewer_frFFTviewer.cpp.o CMakeFiles/LimeSuiteGUI.dir/lms7suiteEvents/lms7suiteEvents.cpp.o CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlQSpark.cpp.o CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnluLimeSDR.cpp.o CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlUltimateEVB.cpp.o CMakeFiles/LimeSuiteGUI.dir/kissFFT/kiss_fft.c.o CMakeFiles/LimeSuiteGUI.dir/windowFunction.cpp.o CMakeFiles/LimeSuiteGUI.dir/boards_wxgui/pnlLimeSDR.cpp.o  -o ../bin/LimeSuiteGUI -rdynamic libLimeSuite.so.17.02.0 oglGraph/liboglGraph.a -L/usr/lib/aarch64-linux-gnu -pthread -lwx_gtk2u_richtext-3.0 -lwx_baseu_xml-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_aui-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu-3.0 -lwx_gtk2u_gl-3.0 -lsqlite3 -lusb -lGL -lwx_gtk2u_richtext-3.0 -lwx_baseu_xml-3.0 -lwx_gtk2u_html-3.0 -lwx_gtk2u_xrc-3.0 -lwx_gtk2u_adv-3.0 -lwx_gtk2u_aui-3.0 -lwx_gtk2u_core-3.0 -lwx_baseu-3.0 -lwx_gtk2u_gl-3.0 -Wl,-rpath,/home/kc7noaa/Downloads/LimeSuite/build/src: 
libLimeSuite.so.17.02.0: undefined reference to `libusb_control_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_get_device_descriptor'
libLimeSuite.so.17.02.0: undefined reference to `libusb_strerror'
libLimeSuite.so.17.02.0: undefined reference to `libusb_set_debug'
libLimeSuite.so.17.02.0: undefined reference to `libusb_handle_events_timeout_completed'
libLimeSuite.so.17.02.0: undefined reference to `libusb_open_device_with_vid_pid'
libLimeSuite.so.17.02.0: undefined reference to `libusb_submit_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_close'
libLimeSuite.so.17.02.0: undefined reference to `libusb_init'
libLimeSuite.so.17.02.0: undefined reference to `libusb_free_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_detach_kernel_driver'
libLimeSuite.so.17.02.0: undefined reference to `libusb_open'
libLimeSuite.so.17.02.0: undefined reference to `libusb_get_device_speed'
libLimeSuite.so.17.02.0: undefined reference to `libusb_cancel_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_release_interface'
[... cut ...]
src/CMakeFiles/LimeSuiteGUI.dir/build.make:1528: recipe for target 'bin/LimeSuiteGUI' failed
make[2]: *** [bin/LimeSuiteGUI] Error 1
make[2]: Leaving directory '/home/kc7noaa/Downloads/LimeSuite/build'
CMakeFiles/Makefile2:172: recipe for target 'src/CMakeFiles/LimeSuiteGUI.dir/all' failed
make[1]: *** [src/CMakeFiles/LimeSuiteGUI.dir/all] Error 2
make[1]: Leaving directory '/home/kc7noaa/Downloads/LimeSuite/build'
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
kc7noaa@odroid64:~/Downloads/LimeSuite/build$

guruofquality, 2017-02-25:
@KC7NOA Oh I was looking for the part of the output where libusb was linked in to libLimeSuite, just to confirm that was happening. Perhaps could I get the whole verbose in a pastebin or something, I realize it might be too much to dump into github. Thanks

KC7NOA, 2017-02-25:
easy peasy ... pastebin is always there ... posted to last one week ...

http://pastebin.com/UJrtd0Ae

guruofquality, 2017-02-25:
aha it should say ` -pthread -lsqlite3 -lusb-1.0 -lusb-1.0` but instead it shows `-pthread -lsqlite3 -lusb -lusb ` which is the older libusb library. But strangely cmake is showing us that it found the correct libusb1.0 library: `USB_LIBRARIES: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so`. So the error makes sense, but I'm at a loss for as to why/how cmake has done this.

It should turn that library path into a linker argument and I guess that is not happening that way it should. Perhaps its a bug in the version of cmake thats installed. This is a suggested change to get you over this hump (hopefully). And if it doesnt, maybe we will learn a little more:
```
diff --git a/src/ConnectionSTREAM/CMakeLists.txt b/src/ConnectionSTREAM/CMakeLists.txt
index 1638b6d..9f49c6a 100644
--- a/src/ConnectionSTREAM/CMakeLists.txt
+++ b/src/ConnectionSTREAM/CMakeLists.txt
@@ -11,7 +11,7 @@ elseif(UNIX)
     find_package(libusb-1.0)
     set(USB_FOUND ${LIBUSB_1_FOUND})
     set(USB_INCLUDE_DIRS ${LIBUSB_1_INCLUDE_DIRS})
-    set(USB_LIBRARIES ${LIBUSB_1_LIBRARIES})
+    set(USB_LIBRARIES "-lusb-1.0")
 endif()
 
 message(STATUS "USB_INCLUDE_DIRS: ${USB_INCLUDE_DIRS}")
```

KC7NOA, 2017-02-26:
Ok ... im not home right now ... ill install cmake from git rather than repository ... and see if there is a change, then move to the next step

KC7NOA, 2017-02-26:
This didn't come in when i issued a git pull .. is it not pushed?

KC7NOA, 2017-02-26:
Made the change .... 
Though , shouldnt the change been made to ConnectionNovenaRF7 , NovenaRF7 Connection support and not ConnectionSTREAM_UNITE ? --

KC7NOA, 2017-02-28:
Working with gqrx so i guess the change worked .. not sure what is wrong with the linker

guruofquality, 2017-03-01:
I was rereading this, I missed that the Odroid-C2(64bit) cmake actually got it wrong in finding libusb-1.0:
```
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/aarch64-linux-gnu/libusb.so
```

The find script for this has a backup name of "usb" when it cant find "usb-1.0": https://github.com/myriadrf/LimeSuite/blob/master/cmake/Modules/Findlibusb-1.0.cmake#L62 Which I think is part of the problem.

* One way around this is to pass the full libusb-1.0.so path via `-DLIBUSB_1_LIBRARY=/usr/blah.../libusb-1.0.so`
* I'm curious if the script finds libusb-1.0 automatically if you delete "usb" from the line with `NAMES usb-1.0 usb` because I dont think there is any case where the newer library it called usb without the version number (except freebsd). I'm thinking along the lines of a permanent change, just curious how it works out for you.

KC7NOA, 2017-03-03:
Ok .. iv recently did a git pull .. and i don't think this file was part of the change ...

i changed LimeSuite/src/ConnectionSTREAM/CMakeLists.txt  back towhat it was before ...

then changed LimeSuite/cmake/Modules/Findlibusb-1.0.cmake 
line 64 to 

usb-1.0

Building ---

KC7NOA, 2017-03-03:
Nope ... 

libLimeSuite.so.17.02.0: undefined reference to `libusb_get_device_descriptor'
libLimeSuite.so.17.02.0: undefined reference to `libusb_claim_interface'
libLimeSuite.so.17.02.0: undefined reference to `libusb_bulk_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_close'
libLimeSuite.so.17.02.0: undefined reference to `libusb_kernel_driver_active'
libLimeSuite.so.17.02.0: undefined reference to `libusb_release_interface'
libLimeSuite.so.17.02.0: undefined reference to `libusb_set_debug'
libLimeSuite.so.17.02.0: undefined reference to `libusb_init'
libLimeSuite.so.17.02.0: undefined reference to `libusb_alloc_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_get_active_config_descriptor'
collect2: error: ld returned 1 exit status
src/CMakeFiles/LimeUtil.dir/build.make:123: recipe for target 'src/LimeUtil' failed
make[2]: *** [src/LimeUtil] Error 1
CMakeFiles/Makefile2:209: recipe for target 'src/CMakeFiles/LimeUtil.dir/all' failed
make[1]: *** [src/CMakeFiles/LimeUtil.dir/all] Error 2
[ 44%] Linking CXX static library liboglGraph.a
[ 44%] Built target oglGraph
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
kc7noaa@odroid64:~/Downloads/LimeSuite/build$

KC7NOA, 2017-03-03:
Paste bin again ...

http://pastebin.com/qwPNb3fM

KC7NOA, 2017-03-04:
Tried .... make-4.1 and cmake-3.8 ... same
Wonder if its something to do with the location arm puts libusb-1.0 ... dont know.

libLimeSuite.so.17.02.0: undefined reference to `libusb_alloc_transfer'
libLimeSuite.so.17.02.0: undefined reference to `libusb_free_device_list'
libLimeSuite.so.17.02.0: undefined reference to `libusb_error_name'
libLimeSuite.so.17.02.0: undefined reference to `libusb_reset_device'
libLimeSuite.so.17.02.0: undefined reference to `libusb_claim_interface'
libLimeSuite.so.17.02.0: undefined reference to `libusb_free_config_descriptor'
collect2: error: ld returned 1 exit status
src/CMakeFiles/LimeSuiteGUI.dir/build.make:1528: recipe for target 'bin/LimeSuiteGUI' failed
make[2]: *** [bin/LimeSuiteGUI] Error 1
CMakeFiles/Makefile2:172: recipe for target 'src/CMakeFiles/LimeSuiteGUI.dir/all' failed
make[1]: *** [src/CMakeFiles/LimeSuiteGUI.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
kc7noaa@odroid64:~/Downloads/LimeSuite/build$ cmake --version
cmake version 3.8.20170304-gd96d97

CMake suite maintained and supported by Kitware (kitware.com/cmake).
kc7noaa@odroid64:~/Downloads/LimeSuite/build$ make -v
GNU Make 4.1
Built for aarch64-unknown-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
kc7noaa@odroid64:~/Downloads/LimeSuite/build$

romeojulietthotel, 2017-03-04:
Make sure after you modify cmake files that you clean out your build directory. 

```
cd ./build
rm CMakeCache.txt
cmake ../ 2>&1 | tee cmake.log

```
sometimes I even do

```
cd ./build
rm -rfv CMake*
cmake ../ 2>&1 | tee cmake.log

```

Also with cmake you want the log because it can tell you things that might surprise you.

This bug should have odroid in the title just so it's known.

KC7NOA, 2017-03-05:
Something changed ... might be from doing 
rm CmakeCasche.txt or something else ... i just dont know .. but ConnectionNovenaRF7 is in the enabled list where previously it was not.

######################################################
-- ## LimeSuite enabled features
-- ######################################################
-- 
 * LimeSuiteHeaders, The lime suite headers
 * LimeSuiteLibrary, The lime suite library
 * ConnectionSTREAM, STREAM Connection support
 * ConnectionNovenaRF7, NovenaRF7 Connection support
 * Connection_uLimeSDR, uLimeSDR Connection support
 * ConnectionXillybus, PCIE Xillybus Connection support
 * LimeSuiteGUI, GUI Application for LimeSuite
 * LimeUtilCommand, Command line device discovery utility
 * LimeSuiteExamples, LimeSuite library API examples
 * SoapySDRLMS7, SoapySDR bindings for LMS7
 * LimeSuiteDesktop, LimeSuite freedesktop integration

-- ######################################################
-- ## LimeSuite disabled features
-- ######################################################
-- 
 * ConnectionEVB7COM, EVB+COM Connection support
 * ConnectionSTREAM_UNITE, STREAM+UNITE Connection support
 * LimeSuiteUtilities, Additional utility programs for testing
 * LimeSuiteTests, Download gtest and build test application

going to

make VERBOSE=1 -j4 in a bit and will report back ...

KC7NOA, 2017-03-05:
ok, well build and installed ....
not sure if removing the usb in line 

then changed LimeSuite/cmake/Modules/Findlibusb-1.0.cmake
line 64

guruofquality, 2017-03-06:
BTW, I changed this library search name to usb-1.0, it switches to usb under freebsd. All of the other build projects with libusb that I could find also do this: https://github.com/myriadrf/LimeSuite/commit/827cb0402b7b606505e58ac637876a1bc6f19958 So chances are you just need to delete the old CMakeCache.txt and everything should work. We can reopen if this isnt the fix.

closed 2017-03-06 by guruofquality

KC7NOA, 2017-03-11:
Yes .. this did fix it ... 
I have also done another git pull and built with no issues now.

mfoco, 2019-02-28:
I found this old thread that might be related to my current problem: I found out that libusb.so is *not* an .so at all: running `ldd libusb.so` reports the error "not a dynamic executable" for aarch64. I'm trying to get in touch with the maintainers of libusb package for clarifications.