myriadrf/LimeSuite #101: Cant tune below 14Mhz now after yesterday git pull

KC7NOA, 2017-03-03:
Wish i could bring up the history to see what was changed with my last git pull ... 
but 2 days ago i could tune down below 2Mhz and now i cant ...

not sure if the change is in some way affecting the way GQRX behaves ...

KC7NOA, 2017-03-03:
At an input rate of 5Msps and 1Msps it would work ..
10Msps works with decimation of 16 .. -- not 100% what is the lowest that functions ...

No changes to gqrx in over a week and same with SoapySDR.

guruofquality, 2017-03-06:
I think its a matter of the lower sample rate; because the decimation chain can only do at most 32, a 1 Msps rate means that the ADC rate will be 32 Msps. So when the LO tunes down to 30 MHz, we can only downshift by an additional 16 MHz, which is where the 14 MHz comes from. With 5 Msps, thats an ADC rate of 160, which should be plenty of room to compensate for the entire 0 to 30 MHz range.

IgnasJarusevicius, 2018-06-22:
It seems like this has been answered/explained by guruofquality

closed 2018-06-22 by IgnasJarusevicius