myriadrf/LimeSuite #104: [ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.5-2/libLMS7Support.so)

maswell, 2017-03-08:
Hi, 
I encountered some problems when I ran GQRX as well as gr-osmosdr module in gnuradio

```
$ gqrx
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.001-0-unknown

Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
FM demod gain: 1.52789
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/maswell/.config/gqrx/bookmarks.csv
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.5-2/libLMS7Support.so)
  duplicate entry for lime (/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.5-2/libLMS7Support.so)
getDeviceList : Available input devices:
    0 : "RFSPACE SDR-IQ Receiver"
    1 : "RFSPACE SDR-IP Receiver"
    2 : "RFSPACE NetSDR Receiver"
    3 : "RFSPACE Cloud-IQ Receiver"
    4 : "LimeSDR-USB [USB 3.0] 9060XXXXXXXX"
    5 : "RTL-SDR Spectrum Server"
    6 : "Red Pitaya Transceiver Server"
    7 : "Complex Sampled (IQ) File"
```
How can i fix it,
Thx

guruofquality, 2017-03-08:
It looks like you installed LimeSuite both from source and from binaries (the myriadrf PPA I presume)? You could end up seeing other problems aside from just some error prints. I recommend only choosing one of those options and uninstalling the other.

maswell, 2017-03-09:
thx for guruofquality's help.
Actually, I did install limesuit twice(source and apt-get).
I ran the following command to solve my problem.
```
sudo apt-get remove limesuite limesuite-udev limesuite-images
sudo apt-get remove soapysdr soapysdr-module-lms7
sudo apt autoremove
```
thx again.

closed 2017-03-09 by maswell