myriadrf/LimeSuite #107: Running rx_sdr just dies with a SEGV

darkstar007, 2017-03-11:
If I run the command "rx_sdr -d driver=lime -F CS16  -f 225.25e6  zxxx.cs16" it just dies with a SEGV:

```
CGEN: Freq=262.144 MHz, VCO=2.62144 GHz, INT=84, FRAC=349525, DIV_OUTCH_CGEN=4
ConnectionSTREAM::ConfigureFPGA_PLL(tx=2.048MHz, rx=2.048MHz)
Sampling at 2048000 S/s.
[INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 225.25 MHz)
SetFrequency using cache values vco:2, csw:172
[WARNING] GetDC_IQ_Interp(225.25 MHz, ch=0, tx=0): no matches between [224.25, 226.25] MHz
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 6.10352e-07 MHz)
Tuned to 225250000 Hz.
Reading samples in sync mode...
Segmentation fault
'''

I have to use rx_sdr as no command line tools seem to be supplied by limesuite to do a simple dump of samples to a file - obviously this introduces a whole bunch of additional packages which complicates everything.

So, running in the debugger I get this:

```
(gdb) r -d driver=lime -F CS16  -f 225.25e6  zxxx.cs16
Starting program: /usr/local/bin/rx_sdr -d driver=lime -F CS16  -f 225.25e6  zxxx.cs16
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
linux; GNU C++ version 6.2.0 20161103; Boost_106200; UHD_003.009.005-0-unknown

[New Thread 0x7fffea969700 (LWP 1718)]
[Thread 0x7fffea969700 (LWP 1718) exited]
[New Thread 0x7ffff7e86700 (LWP 1719)]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
[Thread 0x7ffff7e86700 (LWP 1719) exited]
[New Thread 0x7fffea969700 (LWP 1720)]
[Thread 0x7fffea969700 (LWP 1720) exited]
[New Thread 0x7fffea969700 (LWP 1721)]
[Thread 0x7fffea969700 (LWP 1721) exited]
[New Thread 0x7fffea969700 (LWP 1722)]
[New Thread 0x7fffe9905700 (LWP 1723)]
[New Thread 0x7fffe9104700 (LWP 1724)]
[New Thread 0x7fffe8903700 (LWP 1725)]
[New Thread 0x7fffe3fff700 (LWP 1726)]
[New Thread 0x7fffe37fe700 (LWP 1727)]
[Thread 0x7fffe37fe700 (LWP 1727) exited]
[New Thread 0x7fffe37fe700 (LWP 1728)]
[Thread 0x7fffe37fe700 (LWP 1728) exited]
[New Thread 0x7fffe37fe700 (LWP 1729)]
Found 4 device(s):
[Thread 0x7fffe37fe700 (LWP 1729) exited]
  0: default_input=True, default_output=True, device_id=0, driver=audio, label=hw:HDA Intel PCH,0, 
  1: default_input=False, default_output=False, device_id=1, driver=audio, label=hw:HDA Intel PCH,2, 
  2: default_input=False, default_output=False, device_id=3, driver=audio, label=default, 
  3: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9060A02593414, media=USB 3.0, module=STREAM, name=LimeSDR-USB, serial=0009060A02593414, 

verbose_device_search(driver=lime)
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9060A02593414'
Estimated reference clock 30.7196 MHz
Selected reference clock 30.720 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] Init LMS7002M(0)
LMS7002M values cache at /home/matt/.limesuite/LMS7002M_cache_values.db
[INFO] Ver=7, Rev=1, Mask=1
[INFO] LMS7002M calibration values caching Enable
CGEN: Freq=80 MHz, VCO=2.56 GHz, INT=82, FRAC=349525, DIV_OUTCH_CGEN=15
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 0, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 0, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 0, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 0, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 0, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 0, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=182, N=7, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 100.4 (steps 58)
[INFO] SoapyLMS7::setSampleRate(Tx, 0, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 95.2 (steps 110)
[INFO] SoapyLMS7::setBandwidth(Rx, 0, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 0, 30 MHz)
[INFO] Tx Filter calibrated from cache
[INFO] SoapyLMS7::setFrequency(Rx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setFrequency(Tx, 1, BB, 0 MHz)
[INFO] SoapyLMS7::setAntenna(Rx, 1, LNAL)
[INFO] SoapyLMS7::setAntenna(Tx, 1, BAND1)
[INFO] SoapyLMS7::setGain(Rx, 1, PGA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, LNA, 0 dB)
[INFO] SoapyLMS7::setGain(Rx, 1, TIA, 0 dB)
[INFO] SoapyLMS7::setGain(Tx, 1, PAD, -50 dB)
[INFO] SoapyLMS7::setSampleRate(Rx, 1, 10 MHz), CGEN=80 MHz, ADC=20 MHz, decim=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=20MHz, rx=10MHz)
----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=182, N=7, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 100.4 (steps 58)
[INFO] SoapyLMS7::setSampleRate(Tx, 1, 10 MHz), CGEN=80 MHz, DAC=20 MHz, interp=2
ConnectionSTREAM::ConfigureFPGA_PLL(tx=10MHz, rx=10MHz)

----- FPGA PLL #1 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 114.2 (steps 132)
----- FPGA PLL #0 CONFIG -----
M=156, N=3, Fvco=1040.000 MHz
Configured phase: 0.0 (steps 0)
Configured phase: 95.2 (steps 110)
[INFO] SoapyLMS7::setBandwidth(Rx, 1, 30 MHz)
[INFO] Rx Filter calibrated from cache
[INFO] SoapyLMS7::setBandwidth(Tx, 1, 30 MHz)
[INFO] Tx Filter calibrated from cache
Using device LimeSDR-USB: boardSerialNumber=0x9060a02593414 firmwareVersion=3 hardwareVersion=4 protocolVersion=1 
Found 6 antenna(s): NONE LNAH LNAL LNAW LB1 LB2 
Found 3 gain(s): LNA TIA PGA 
Found 2 frequencies: RF BB 
Found 2 sample rates: 100000 60000000 
Found 0 bandwidths: 
Using output format: CS16 (input format CS16)
[INFO] SoapyLMS7::setSampleRate(Rx, 0, 2.048 MHz), CGEN=262.144 MHz, ADC=65.536 MHz, decim=32
CGEN: Freq=262.144 MHz, VCO=2.62144 GHz, INT=84, FRAC=349525, DIV_OUTCH_CGEN=4
ConnectionSTREAM::ConfigureFPGA_PLL(tx=2.048MHz, rx=2.048MHz)
Sampling at 2048000 S/s.
[INFO] SoapyLMS7::setFrequency(Rx, 0, RF, 225.25 MHz)
SetFrequency using cache values vco:2, csw:172
[WARNING] GetDC_IQ_Interp(225.25 MHz, ch=0, tx=0): no matches between [224.25, 226.25] MHz
[INFO] SoapyLMS7::setFrequency(Rx, 0, BB, 6.10352e-07 MHz)
Tuned to 225250000 Hz.
Reading samples in sync mode...
[New Thread 0x7fffe37fe700 (LWP 1730)]
[New Thread 0x7fffcffff700 (LWP 1731)]

Thread 1 "rx_sdr" received signal SIGSEGV, Segmentation fault.
0x00007ffff73db854 in clock_gettime () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) where
#0  0x00007ffff73db854 in clock_gettime () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff6e1262e in std::chrono::_V2::system_clock::now() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007ffff5e868a5 in std::condition_variable::wait_for<long, std::ratio<1l, 1000l> > (__rtime=..., __lock=..., this=0x5555558029c8)
    at /usr/include/c++/6/condition_variable:138
#3  lime::RingFIFO::pop_samples (flags=0x7fffffffe288, timeout_ms=1000, timestamp=0x7fffffffe280, channelsCount=1 '\001', samplesCount=262144, buffer=0x7ffff7f32010, 
    this=0x555555802980) at ./src/protocols/fifo.h:135
#4  lime::ILimeSDRStreaming::StreamChannel::Read (this=<optimized out>, samples=0x7ffff7f32010, count=262144, meta=0x7fffffffe280, timeout_ms=1000)
    at ./src/protocols/ILimeSDRStreaming.cpp:395
#5  0x00007ffff5e855df in lime::ILimeSDRStreaming::ReadStream (this=<optimized out>, streamID=<optimized out>, buffs=<optimized out>, length=<optimized out>, 
    timeout_ms=<optimized out>, metadata=...) at ./src/protocols/ILimeSDRStreaming.cpp:100
#6  0x00007ffff61071c8 in SoapyLMS7::readStream (this=0x55555580c440, stream=0x55555576d120, buffs=0x7fffffffe400, numElems=262144, flags=@0x7fffffffe3fc: 0, 
    timeNs=@0x7fffffffe408: 0, timeoutUs=1000000) at ./SoapyLMS7/Streaming.cpp:247
#7  0x00007ffff7bc47a4 in SoapySDRDevice_readStream () from /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.5-2
#8  0x0000555555555f7e in main ()
(gdb)
```

So it appears its an issue with clock_gettime called (indirectly) from liblimesuite?

This is with the latest version of limesuite checked out an hour or so ago from github (and similarly for rx_sdr).

Thanks,

Matt

guruofquality, 2017-03-11:
Bug fix in rx utils. Seems there was an allocation issue. The fix is on master; can you test it as well? https://github.com/rxseger/rx_tools/commit/886cd3a058dd7fb967513ddc978f6ad13e0f7445

darkstar007, 2017-03-11:
It certainly runs now! Now just need to add an antenna and look at the data....

guruofquality, 2017-03-13:
Looks to be resolved. Thanks! https://github.com/rxseger/rx_tools/issues/47

closed 2017-03-13 by guruofquality