myriadrf/LimeSuite #115: Segmentation fault when using uhd_siggen

ashley-b, 2017-04-03:
When i run the following command I get this error
uhd_siggen -f100e+6
Segmentation fault (core dumped)

When I debug it I find
lime::ILimeSDRStreaming::StreamChannel::Write
argument samples is pointing to NULL.

gdb output

0x00007fffd9401715 in lime::ILimeSDRStreaming::StreamChannel::Write (this=0x7fffc0000be0, samples=0x0, count=1, meta=0x7fffd37fcf30, timeout_ms=1000)
    at LimeSuite/src/protocols/ILimeSDRStreaming.cpp:413
413	            samplesShort[i] = samplesFloat[i]*2047;
(gdb) backtrace 
#0  0x00007fffd9401715 in lime::ILimeSDRStreaming::StreamChannel::Write (this=0x7fffc0000be0, samples=0x0, count=1, meta=0x7fffd37fcf30, 
    timeout_ms=1000) at LimeSuite/src/protocols/ILimeSDRStreaming.cpp:413
#1  0x00007fffd9400078 in lime::ILimeSDRStreaming::WriteStream (this=0x555556487190, streamID=140736414616544, buffs=0x0, length=1, timeout_ms=1000, 
    metadata=...) at LimeSuite/src/protocols/ILimeSDRStreaming.cpp:114
#2  0x00007fffd97d8a12 in SoapyLMS7::writeStream (this=0x55555649d660, stream=0x7fffc0000980, buffs=0x7fffc0000a10, numElems=1, 
    flags=@0x7fffd37fd028: 0, timeNs=150000000, timeoutUs=1000000) at LimeSuite/SoapyLMS7/Streaming.cpp:333
#3  0x00007fffe8f2fc01 in UHDSoapyTxStream::send(uhd::ref_vector<void const*> const&, unsigned long, uhd::tx_metadata_t const&, double) ()
   from /usr/local/lib64/uhd/modules/libsoapySupport.so
#4  0x00007fffefa63f68 in gr::uhd::usrp_sink_impl::start() () from /usr/local/lib64/libgnuradio-uhd-3.7.11.so.0.0.0
#5  0x00007fffeec842e2 in gr::block_executor::block_executor(boost::shared_ptr<gr::block>, int) ()
   from /usr/local/lib64/libgnuradio-runtime-3.7.11.so.0.0.0
#6  0x00007fffeecd75b3 in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, int) ()
   from /usr/local/lib64/libgnuradio-runtime-3.7.11.so.0.0.0
#7  0x00007fffeecc9c82 in boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke(boost::detail::function::function_buffer&) () from /usr/local/lib64/libgnuradio-runtime-3.7.11.so.0.0.0
#8  0x00007fffeec6b832 in boost::detail::thread_data<boost::function0<void> >::run() () from /usr/local/lib64/libgnuradio-runtime-3.7.11.so.0.0.0
#9  0x00007fffed3d4146 in thread_proxy () from /lib64/libboost_thread.so.1.60.0
#10 0x00007ffff77e06ca in start_thread () from /lib64/libpthread.so.0
#11 0x00007ffff6e0af7f in clone () from /lib64/libc.so.6

jocover, 2017-04-04:
bug on https://github.com/pothosware/SoapyUHD/blob/master/UHDSoapyDevice.cpp#L718
change to this code
` if (nsamps_per_buff == 0) return 0;
`

guruofquality, 2017-04-04:
Good find. some of the uhd apps use zero length submissions with end of burst flag to mark the end of burst. I think that I put that line there to pad out the zero length to support some hardware which could not handle the empty buffer submission. However, the buffers being passed in are null pointers, and we cannot index the first sample of a null pointer. I will remove this line. Its the job of the hardware drivers/ modules that support end of burst to deal with length zero appropriately.

ashley-b, 2017-04-21:
Thanks for the fix

closed 2017-04-21 by ashley-b