myriadrf/LimeSuite #118: LMS7002M::TuneRxFilterWithCaching() not working properly

daniestevez, 2017-04-08:
I have been testing different bandwidths for the LimeSDR lowpass filter using GQRX and there is a bug for some particular bandwidths, where the filter is not setup properly. The cutoff frequency is sometimes completely wrong or the passband has a lot of variation (5dB or so).

For an example, see this tweet and some of my replies: https://twitter.com/ea4gpz/status/849373269312643080

I have been able to track down the issue to `LMS7002M::TuneRxFilterWithCaching()`. If I replace this function by `LMS7002M::TuneRxFilter()` then the lowpass filter works as expected.

daniestevez, 2017-04-16:
Workaround for this bug is here:

https://github.com/daniestevez/LimeSuite/commit/fe9a09a4ff5b79b523c188dbd76390b31e2f2ef0

guruofquality, 2017-05-22:
Fixed an issue with the register cache: https://github.com/myriadrf/LimeSuite/commit/77d3689f9d13c923a49bba92a9a2ae7f26ede9ab

The old ~/.limesuite/LMS7002M_cache_values.db will have to be deleted

daniestevez, 2018-01-05:
It seems that this has eventually been fixed.

closed 2018-01-05 by daniestevez