myriadrf/LimeSuite #119: wxpython not detected ( wxWidgets??)

KC7NOA, 2017-04-10:
Failed to set RX input rate to 2.5e+06
Your device may not be working properly.

Decimation: 8
  stage: 1  ratio: 8
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya

FATAL: No supported devices found to pick from.

kc7noa@kc7noaavera:~/Downloads/LimeSuite/buildme$ cmake ../
-- SQLITE3: TRUE
-- SQLITE3_INCLUDE_DIRS: /usr/include
-- SQLITE3_LIBRARIES: /usr/lib/i386-linux-gnu/libsqlite3.so
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/i386-linux-gnu/libusb-1.0.so
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/i386-linux-gnu/libusb-1.0.so
-- USB_INCLUDE_DIRS: /usr/include/libusb-1.0
-- USB_LIBRARIES: /usr/lib/i386-linux-gnu/libusb-1.0.so
-- Could NOT find wxWidgets (missing:  wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS)
-- wxWidgets_DEFINITIONS:
-- wxWidgets_INCLUDE_DIRS:
-- wxWidgets_LIBRARIES:
-- GNUPlot Found
-- 
-- ######################################################
-- ## LimeSuite enabled features
-- ######################################################
-- 
 * LimeSuiteHeaders , The lime suite headers
 * LimeSuiteLibrary , The lime suite library
 * ConnectionSTREAM , STREAM Connection support
 * Connection_uLimeSDR , uLimeSDR Connection support
 * ConnectionXillybus , PCIE Xillybus Connection support
 * LimeUtilCommand , Command line device discovery utility
 * LimeSuiteExamples , LimeSuite library API examples
 * SoapySDRLMS7 , SoapySDR bindings for LMS7
 * LimeSuiteDesktop , LimeSuite freedesktop integration

-- ######################################################
-- ## LimeSuite disabled features
-- ######################################################
-- 
 * ConnectionEVB7COM , EVB+COM Connection support
 * ConnectionSTREAM_UNITE , STREAM+UNITE Connection support
 * ConnectionNovenaRF7 , NovenaRF7 Connection support
 * LimeSuiteGUI , GUI Application for LimeSuite
 * LimeSuiteUtilities , Additional utility programs for testing
 * LimeSuiteTests , Download gtest and build test application

-- Install prefix: /usr/local
-- Build timestamp: 2017-04-10
-- Lime Suite version: 17.03.0-g2c225deb
-- ABI/so version: 17.03-1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kc7noa/Downloads/LimeSuite/buildme
kc7noa@kc7noaavera:~/Downloads/LimeSuite/buildme$ 

This probably needs slightly better documentation to resolve in the makefile when its not found ...
Like which wxpython and perhapse where to get it ...

I seem to remember resolving this issue by downloading wxpython2.8 from some where ... cant remember and ill have time later to try to figure out how i got around this before ....

Im trying out an OLD laptop AMD 2.2Ghz with Sis GPU ... Ubuntu 16.04.1

guruofquality, 2017-04-14:
> I seem to remember resolving this issue by downloading wxpython2.8 from some where ... cant remember and ill have time later to try to figure out how i got around this before ....
Im trying out an OLD laptop AMD 2.2Ghz with Sis GPU ... Ubuntu 16.04.1

Did you run the apt-get instructions from the wiki? http://wiki.myriadrf.org/Lime_Suite#Ubuntu

Specifically these:
```
#install graphics dependencies
sudo apt-get install libwxgtk3.0-dev freeglut3-dev
```

It should readily find the regular wx development files from ubuntu's package manager.

KC7NOA, 2017-04-15:
I did .....

Though maybe not before building gr-oscom .... it did not list soapy as a suported device so i went back to rebuild it .....

I did see soapysdr durring the cmake phase so maybe it will work once make has finishrd

KC7NOA, 2017-04-15:
ok ... i think im on a track .. i don't know why ... but soapy is not in the list when starting gqrx
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya

When i run gqrx on my Odroid-X2 soapy IS in the list .... i need to examine the cmake output of gr-osmosdr more.

SoapySDRUtil lists the lime and SoapySDRUtil --check=lime reports a positive output ....

So i guess this isnt a LimeSuite issue ...

KC7NOA, 2017-04-16:
Still the same ....

gitmodimo, 2017-04-30:
I had the same thing. Almost filed bug report for cmake.  Problem is repeatable when you try to run cmake twice I the same directory. Cleaning builddir fixes the problem. Probably cmake script needs  exclusion for builddir or something.

guruofquality, 2017-05-22:
Cool. So it seems to be a caching issue. wx deps should be installed before cmake is run. Looks like a cmake bug with their wx widgets find script. Closing

closed 2017-05-22 by guruofquality