myriadrf/LimeSuite #120: Hardware DC removal not working with SoapyLMS7

daniestevez, 2017-04-16:
The problem seems to be that `LMS7002M::GetRxDCRemoval()` always returns false when it is called by SoapyLMS7, thus tricking it into disabling DCRemoval (which should be enabled by default).

The following is a workaround for this bug:
https://github.com/daniestevez/LimeSuite/commit/9fe914f6d9edc36675e79e34efe985c318aa99a2

Results after the workaround:
https://t.co/SeLkb7UGrG

Before the workaround you have a huge DC spike (on the lowest level, `DC_BYP_RXTSP` gets set to 1).

guruofquality, 2017-04-16:
Thanks for the find. It looks like the wrong register was overwritten due to a typo: https://github.com/myriadrf/LimeSuite/commit/2c3676c8084922a7013e389c9a5e605c0b3d3838

daniestevez, 2017-04-16:
Thanks. That solves it.

closed 2017-04-16 by daniestevez