myriadrf/LimeSuite #121: No API in LimeSuite to disable NCO

f4exb, 2017-04-20:
Hello,

I haven't seen anywhere in the API (lime/LimeSuite.h) a way to disable the NCO once it is enabled with LMS_SetNCOFrequency. There is no explicit enable boolean variable (unlike LMS_SetGFIRLPF for example) nor setting the NCO frequency to zero does that apparently.

If I got it right this is done in LMS7_Device class simply with this instruction: Modify_SPI_Reg_bits(LMS7param(CMIX_BYP_RXTSP), 1);

Alas there is no way to activate it unless you init the device.

It would be nice to have a way to just disable the NCO via an API call.

Best regards.

f4exb, 2017-04-20:
My bad... LMS_SetNCOIndex just does that

IgnasJarusevicius, 2017-04-20:
Hi,
thanks for observation, I will update LMS_SetNCOIndex() description.

You can also do it the same way as from LMS7_Device class:
LMS_WriteParam(device,LMS7param(CMIX_BYP_RXTSP),1);

f4exb, 2017-04-20:
Maybe you want to make this more explicit or have a dedicated API

f4exb, 2017-04-20:
Yes using LMS_WriteParam is probably more simple and straightforward

closed 2017-04-20 by f4exb