myriadrf/LimeSuite #123: Error setting sample rate to 60 MHz (and 61.44 MHz sample rate missing in provided ranges)

xmikos, 2017-04-26:
When trying to set sample rate to 60 MHz, I get this error:

```
[ERROR] setSampleRate(Rx 60 MHz, Tx 60 MHz) Failed -- no common clock rate
```

Setting sample rate to 61.44 MHz works, but strange is that this sample rate is not provided by `listSampleRates` or `getSampleRateRange`. Ranges returned by both of these methods are only 100 kHz - 60 MHz.

So there are two errors:
1. setting sample rate to 60 MHz (which is provided in list of allowed ranges) doesn't work
2. setting sample rate to 61.44 MHz does work, but it isn't provided in list of allowed ranges

guruofquality, 2017-05-03:
You are right, the actual range is a little finer grained. I made this commit to more accurately portray the limits: https://github.com/myriadrf/LimeSuite/commit/7be5179ce76dae6f2f6c096f32d43a966ab996ac

The error message however, is accurate, because there is really not an available clock rate to meet the 60 MHz setting.

Also as of the latest SoapySDR release, listSampleRates is now deprecated in favor of getSampleRateRange() which can show multiple continuous ranges which is indeed the case here.

So I hope that helps. Try it out, if something seems off, lets reopen this.

closed 2017-05-03 by guruofquality

xmikos, 2017-05-04:
Thanks, it works.

I am already using getSampleRateRange in my SimpleSoapy wrapper.

Btw. what will happen if I select 65 MHz sample rate? I believe 61.44 MHz is the highest possible sample rate over USB 3?