myriadrf/LimeSuite #124: LMS7002M::EnableChannel disables TX ports if called to disable RX ports only

wtcom-apieroni, 2017-05-07:
@guruofquality , We verified using our forked soapy wrapper that calling the function works if disabling TX ports and enabling RX ports. Doesn't work if RX ports need to be disabled but TX ports need to be enabled, a side effect is that TX ports stop transmitting. TxPLL still there though.

wtcom-apieroni, 2017-05-08:
We believe the issue is specifically at this lines

https://github.com/myriadrf/LimeSuite/blob/master/src/lms7002m/LMS7002M.cpp#L254
https://github.com/myriadrf/LimeSuite/blob/master/src/lms7002m/LMS7002M.cpp#L255

Those are global power control lines and shouldn't be modified based on enable or disable

IgnasJarusevicius, 2017-05-08:
Hi,
You are correct about those lines. As it is now, it should work when you do disables first and enables later. Looking further, it looks like there are similar issues with A and B channels as well (disabling one channel disables some things that are used by both channels). I will look into this.

wtcom-apieroni, 2017-05-08:
Please let us know when you may have a patch we can test.

IgnasJarusevicius, 2017-05-09:
I have made a fix: af9aff1b. 
Also, on previous LMS7 chip revision (older board than LimeSDR v1.4), DAC does not work when ADC is disabled. So if you have older chip, you should keep RX channel enabled or at least manually enable ADC.

wtcom-apieroni, 2017-05-09:
Thank you Ignas, right now we are having a very hard time trying to validate the TX output path. We are certain that our RX path works fine, so we can use it for debug.

 Can you provide us some guidelines on how to test the TX Path (Registers to tweak or functions to call)

1) By looping back the RX IQ data internally to the LMS without using the FPGA path (internally routed thru the LMS from Rx to Tx)

2) Using the TxTSP generator (how the output signal should look like?)

Regards,

IgnasJarusevicius, 2017-05-09:
1. To set up digital RX to TX loopback using LimeSuiteGUI: https://discourse.myriadrf.org/t/competition-write-a-simple-limesuite-rxtx-example/1117/13
Without GUI, these parameters can be changed using LMS7002M::Modify_SPI_Reg_bits():
Modify_SPI_Reg_bits(LMS7param(TX_MUX), 2);
Modify_SPI_Reg_bits(LMS7param(TXWRCLK_MUX), 2);

2. It same as RxTSP, so you can check how that looks in RX using LimeSuiteGUI. Basically, when DC mode is selected IQ values are constant, and when NCO mode is selected IQ values are sin and cos that produces 8 or 4 points in constellation (depends if TSP_clk/8 or TSP_clk/4 is selected).

wtcom-apieroni, 2017-05-09:
Hi Ignas,

Right now our main issue is that we had a working Tx Path with rev 7.1.0 but new chips are 7.1.1 and we are not be able to send data properly to the chip. Both TX channels reproduce same signal from baseband and we cannot find any information about the new rev mask. We definitely validated the RX path and looping from RX to TX seems to work fine. Any ideas? we did several tests but we can't make both TX paths stop transmitting same signal. TX calibration also fails to obtain good results and we can see high ripple in passband sometimes.

IgnasJarusevicius, 2017-05-10:
Hi,
What board are you using? If RX to TX loopback works fine, it suggest that the problem is in transferring samples to the chip. 

Also, there has been a fix to stop retransmitting old buffer 3bdc6218da0a5, but this should affect both chip revisions.

integretek-hsandlar, 2017-05-10:
Hi Ignas,
We are using a custom board and FPGA solution. We had a rev1 board which included the 7.1.0 chip and was fully functional. We were extremely surprised to find that the same configuration does not work for the 7.1.1 part -- especially since there was no mention from Lime regarding any changes which would impact our design between the two chip revisions.

We did test RX->TX loopback yesterday and saw it working as expected. However, that test is essentially loopback internal to the LMS. During debug with this new chip revision, we've seen that we now need to invert MCLK to get the correct mappings on the LML interface for our capture hardware. We still haven't been able to get the transmit datapath running as expected again, even though it was fully functional in the 7.1.0 part.

After discussing briefly with @guruofquality, it sounds like Lime has made a variety of changes on the LML interface with this new chip revision which does not make it backwards compatible. He specifically noted RX has a swap for I and Q, the RxTSP NCO is sign reversed, and the Rx MCLK is now delayed relative to the samples. He noted that in order to get the 7.1.1 chip working on the LimeSDR he had to empirically sweep the PLL phase and pick the center of the delay region that worked for synchronizing the TX/RX LML interface.

Can you provide any more detail about what Lime has changed in this 7.1.1 revision chip? We haven't seen any updated documentation regarding this change, and obviously it is not backwards compatible. This is costing us substantial time in debug experimenting to try to figure out what changed, and this issue is currently blocking our production shipping.

IgnasJarusevicius, 2017-05-11:
I have pointed this thread to people that work with hardware and today I got an answer that they are already having an email conversation with you. As your current problems are hardware related I think I can not help you any further, so there is no point to continue the discussion here.
I am closing this as initial problem with channel disables seems to be fixed.

closed 2017-05-11 by IgnasJarusevicius

wtcom-apieroni, 2017-05-11:
Thank you for the support