myriadrf/LimeSuite #128: STORE_RX_CORRECTIONS and STORE_TX_CORRECTIONS removed from Settings.cpp

kb1vc, 2017-06-02:
The commit note said: 
commit 28abe7b3e8f0056aac030ea37e0ac6952f0553aa
Author: Josh Blum <josh@joshknows.com>
Date:   Mon May 22 17:48:15 2017 -0700

    remove apply/store corrections
    
    * these were used with the older calibration utility

--------------------

I was using these too.  

In the absence of a specification or reference manual, it is hard to know what features 
are commitments to the programmer and which features will disappear.  

I use the STORE_TX and STORE_RX corrections because I have seen problems (that is, 
dreadful settings) with the built-in calibration routines.  

Please put these keyword selections back.  It is very very hard to write software for a moving
target.  

matt

guruofquality, 2017-06-02:
@kb1vc The corrections cache stuff is in flux because its being changed to match the new corrections implemented in the MCU of the LMS7. Is the use case here to manually setDCOffset/setIQBalance and then store them into the cache?

kb1vc, 2017-06-02:
@guruofquality 
Thanks for the quick reply.  Yes.  I'd like to store them in the cache, as the MCU calculations do not produce adequate sideband/image rejection.  For a test case, there is code at 
git@github.com:kb1vc/SimpleRXTest.git
(do these references work?)
that produces a plot showing the IQ circle from the MCU calibration vs. the IQ circle from a simple setting to <1.0, 1.0e-6>  ... The latter is much closer to ideal.  I've seen varying degrees of disagreement, sometimes the MCU correction is quite bad, others it is just inferior. 

The code requires gnuplot, fftw3f, and boost::format . 

run the do_experiment.sh script and it will produce build/IQPlot.pdf

kb1vc, 2017-06-02:
> _
[image]
_

IgnasJarusevicius, 2018-06-22:
Closing old issues. I think STORE_TX and STORE_RX  will not be returned to SoapySDR. Currently, SoapySDR::Device::readSetting() could probably be used as a workaround to read relevant LMS7 chipsettings ('DCCORRI_TXTSP', 'DCCORRW_TXTSP', ...);

closed 2018-06-22 by IgnasJarusevicius